YoeDistro/yoe-distro #2: add meta-variscite layers

cbrake, 2018-09-24:
Need to clean up issues where these layers don't play nice with other BSP layers.

kraj, 2018-09-24:
Right now we have disabled few BSP layers among them are meta-ti meta-intel and fslc
eventually we should be able to add them provided they are made multi-BSP friendly

otavio, 2018-09-24:
I'd use meta-freescale instead as it has mainline support for this machine and is well maintained.

kraj, 2018-09-24:
@otavio thanks, can you review

https://github.com/YoeDistro/yoe-distro/pull/3/commits/5afe17de1919d49c602f002669f63ef10a062b06

https://github.com/YoeDistro/yoe-distro/pull/3/commits/9330486f2b087948170bd17ea8486c1b57b26107

otavio, 2018-09-24:
It seems good.

kraj, 2018-09-24:
Added few new boards to env

kraj, 2020-03-01:
The layer is already part of master template since d7897e5c

closed 2020-03-01 by kraj