YoeDistro/yoe-distro #4: Consider using O.S. Systems environment setup scripts

otavio, 2018-09-24:
Hello,

Looking at what you are doing, it shows a lot of commonalities with our environment setup scripts. Take a look at:

https://doc.ossystems.com.br/managing-platforms.html#_code_setup_environment_code_hooks

cbrake, 2018-09-24:
thanks @otavio -- will definitely check it out.

otavio, 2018-09-25:
@kraj you expressed a concern about the scripts to use Python on #5. 

From our point of view, it is a good option as BitBake itself already uses Python and it is indeed a build dependency for the host to enable the use of Yocto Project. Another benefit is it can make use of BitBake libraries (parsing for example) when needed. 

Could you explain why do you see it as a problem?

kraj, 2018-09-25:
it is not a problem at all, its just that Yoe's setup is shell based, so any plugins would mostly follow shell scripts too.

otavio, 2018-09-25:
> it is not a problem at all, its just that Yoe's setup is shell based, so any plugins would mostly follow shell scripts too.

Right, but what I propose here is the replacement.

cbrake, 2018-10-02:
@otavio still working through your setup -- been a busy week, so need to spend a little more time on it. What are your thoughts on git submodules vs repo? My thought to date is repo is overkill to manage a half-dozen repos, but I know this is not a popular opinion. Today as I'm looking at the repo manifest format vs .gitmodules, I'm again just seeing unneeded abstraction and complexity.

kraj, 2018-10-02:
another good datapoint should be to do a survey and ask what people will prefer.

cbrake, 2018-10-02:
One target audience for Yoe is the developer who does not have a lot of Embedded Linux experience and may not be familiar with repo or submodules. The embedded linux experts seem to prefer repo (seems every CPU/SOM Yocto BSP uses repo), however I still wonder if that is the best for new users, or product development.

otavio, 2018-10-02:
> @otavio still working through your setup -- been a busy week, so need to spend a little more time on it. What are your thoughts on git submodules vs repo? My thought to date is repo is overkill to [...]

I dislike `git submodule` at it is very easy to be "surprised" by it, it is fragile. On the other hand, `repo` is overkill but easy to understand and with plenty of docs on the Internet due to its Android use.

I did test it all when designing the FSL Community BSP (back in 2011) and decided to go with `repo`.

cbrake, 2018-10-02:
@otavio one "surprise" with git submodules is the "no-branch" issue if you commit changes on a submodule: http://bec-systems.com/site/696/git-submodules-what-to-do-when-you-commit-to-no-branch.  Are there other areas where you think submodules are fragile?

Since 2011, git submodule has added the capability to track branches: http://bec-systems.com/site/1020/git-submodules-can-now-track-branches

Recently, the commands to remove a submodule got a little better, though still somewhat confusing.

otavio, 2018-10-02:
I really don't see why `git submodule` would be better here. The `repo` capability of including manifests and other advanced things are nice, specially when doing extended products on top of one platform.

cbrake, 2018-10-02:
@otavio I don't want to be a pest -- so just tell me to go away if this gets annoying :-)

One requirement for production build systems is to be able to absolutely know what versions of all layers were used for each release. With submodules, we may update a submodule to the latest branch version (git submodule update --remote sources/meta-xyz), then git add/commit/tag. The git hash of the submodule is recorded. How do you accomplish the same thing in repo? Do you need to specify the git hash in the manifest, or how is this typically done?

kraj, 2018-10-02:
@cbrake one can pin SHA in repo manifests see 

https://github.com/Angstrom-distribution/angstrom-manifest/blob/angstrom-v2018.06-sumo/default.xml#L22

Most of endusers are pretty confused with super repository setup, be it managed by repo or git submodules, with repo the android community has pushed some know-how into developers so
folks are riding on that bet, but it has its own problems of taking you far away from git.

git submodules has been crude in part and has improved over time. And there is a very large communities which now use git submodules to manage super repos. 

I think we need to find practical disadvantages of the tools which can hinder people from using it.
my likes/dislikes are secondary enduser is more important.

Some consideration would be.

1. How easy is for them to consume
2. How easy is the flow for them to contribute back
3. How easy is it for them to learn and re-distribute with control.

otavio, 2018-10-02:
> @otavio I don't want to be a pest -- so just tell me to go away if this gets annoying :-)
>
> One requirement for production build systems is to be able to absolutely know what versions of all layers [...]

Yes; we pin the hashes. 

In fact, `repo` itself has a command to do it. See:

```
$: repo manifest -r
```

We did a tool here at O.S. Systems that does all that and tags the manifest when doing a formal release. After that, reverts the pinned hashes so development continues.

cbrake, 2018-10-02:
@otavio when setting up a build using ossystems-yocto-platform, bblayers.conf and local.conf get copied to a temp directory. Do you store local.conf in git for production builds? If so, how?

otavio, 2018-10-02:
> but some things like local.conf, it seems it would be nice if the user could simply edit directly, rather than having to create a hook.

You can edit it, it won't override it if it already exists.

Please give it a try. It has been in use for years and we polished it a long the way.

kraj, 2020-03-01:
I will close this issue, if we need more discussions we can re-open it

closed 2020-03-01 by kraj