YoeDistro/yoe-distro #23: beaglebone amx3-cm3 compile failure

cbrake, 2018-10-07:
MACHINE=beaglebone

```
ERROR: amx3-cm3-1.9.2-r0 do_compile: Function failed: do_compile (log file is located at /scratch/oe/yoe-distro/build/tmp/work/armv7at2hf-neon-yoe-linux-gnueabi/amx3-cm3/1.9.2-r0/temp/log.do_compile.16020)
ERROR: Logfile of failure stored in: /scratch/oe/yoe-distro/build/tmp/work/armv7at2hf-neon-yoe-linux-gnueabi/amx3-cm3/1.9.2-r0/temp/log.do_compile.16020
Log data follows:
| DEBUG: Executing shell function do_compile
|      CC       src/sys_exec/sys_init.o
| cc1: error: -mfloat-abi=hard: selected processor lacks an FPU
| make: *** [Makefile:51: src/sys_exec/sys_init.o] Error 1
| WARNING: /scratch/oe/yoe-distro/build/tmp/work/armv7at2hf-neon-yoe-linux-gnueabi/amx3-cm3/1.9.2-r0/temp/run.do_compile.16020:1 exit 2 from 'make CROSS_COMPILE="arm-yoe-linux-gnueabi-" CC="arm-yoe-linux-gnueabi-gcc  --sysroot=/scratch/oe/yoe-distro/buil
d/tmp/work/armv7at2hf-neon-yoe-linux-gnueabi/amx3-cm3/1.9.2-r0/recipe-sysroot  -mfloat-abi=hard -no-pie -fno-PIE"'
| ERROR: Function failed: do_compile (log file is located at /scratch/oe/yoe-distro/build/tmp/work/armv7at2hf-neon-yoe-linux-gnueabi/amx3-cm3/1.9.2-r0/temp/log.do_compile.16020)
ERROR: Task (/scratch/oe/yoe-distro/sources/meta-ti/recipes-bsp/cm3-pm-firmware/amx3-cm3_git.bb:do_compile) failed with exit code '1'
```

kraj, 2018-10-09:
posted a patch for meta-ti here

https://lists.yoctoproject.org/pipermail/meta-ti/2018-October/011343.html

cbrake, 2018-10-10:
programmed wic image and getting:

```
U-Boot SPL 2018.01-g2cc52408bf (Oct 09 2018 - 21:10:31)
Trying to boot from MMC1

U-Boot SPL 2018.01-g2cc52408bf (Oct 09 2018 - 21:10:31)
Trying to boot from MMC1
```
will try a few other ways to set up SD card ...

cbrake, 2018-10-10:
Looks like it is not getting into u-boot -- verified MLO and u-boot.img are in the SD boot partition, so could be a u-boot build issue, or a SD formatting issue. 

This thread has some info: https://lists.yoctoproject.org/pipermail/meta-ti/2018-September/011179.html

Will into this more later.

cbrake, 2018-10-12:
note to try GCCVERSION_ti-soc = "7.%" if we hit dead-end with GCC8

kraj, 2018-10-16:
this should be working with```yoe/mut``` branch in out meta-ti fork now

cbrake, 2018-10-17:
builds and boots here. Two issues I'm seeing:

* kernel messages are not enabled on serial console
* login over serial console is behaving very strangely (see attached gif screenshot)

[image]

kraj, 2018-10-17:
Can you try ttyS0 and ttyO0 both as SERIAL_CONSOLES in machine conf

cbrake, 2018-10-17:
ignore the serial console behaving strangly comment -- built latest, and that is working. Following change is required in extlinux.conf to get kernel messages on serial console:

```
root@beaglebone:~# more /media/boot/extlinux/extlinux.conf 
default Yocto
label Yocto
   kernel /zImage
   fdtdir /
append root=PARTUUID=1e4730e5-02 rootwait console=ttyO0,115200n8
```

cbrake, 2018-10-18:
@kraj what are your thoughts about adding meta-ti to bblayers.conf, or should we just document that?

kraj, 2018-10-18:
May be not yet, there is one u-boot patch I have sent last night, that needs to be applied. once thats in then we can build images for beaglebone out of box. Hopefully it will be in soon

kraj, 2018-11-07:
I think this should be fixed now with 48662477a3498d553018e9bac4e09793bc3b622b

closed 2018-11-07 by kraj