YoeDistro/yoe-distro #28: systemd support

cbrake, 2018-10-09:
Would like a documented way to switch between sysvinit and systemd, and test on a number of platforms.

cbrake, 2018-10-09:
interesting comparing the disk usage of sysvinit to systemd image for core-image-minimal on rPI2, 4.7MB vs 33MB. Adding up the sizes in the buildhistory file, I get 22MB for systemd image, and 4.3MB for sysvinit, so perhaps part of the issue is file system efficiency with small files or something.

History files are attached -- you can open easily by renaming the file to have a *.csv extension, then running soffice <file.csv>

[package-sizes-systemd.txt](https://github.com/YoeDistro/yoe-distro/files/2460784/package-sizes-systemd.txt)
[package-sizes-sysvinit.txt](https://github.com/YoeDistro/yoe-distro/files/2460785/package-sizes-sysvinit.txt)

cbrake, 2018-10-09:
The sysvinit image has 696 files, and the systemd image has 1,806 files.

kraj, 2018-10-09:
yes ~16M overhead with systemd I see. compare the size of .tar.bz2 rootfs and not wic image. Because wic image will be aligned so there might be padding.

cbrake, 2018-10-09:
I was looking at actual space used on running system using df to get 4.7MB and 33MB.

kraj, 2018-10-09:
you might have to ignore directories like /var/log and /var/tmp since they might contain generated files after boot. Especially the  journal.

cbrake, 2018-10-09:
/var only has 177KB in it -- hmm.

cbrake, 2018-10-09:
closing for now -- re-open if you think there is anything else we need to do for this.

closed 2018-10-09 by cbrake