YoeDistro/yoe-distro #52: Wrong reference to meta-rust

wouterlucas, 2018-11-11:
When I run yoe_setup on a clean setup:

```
wouter@buildserver:~/workspace/rpi/yoe-distro$ yoe_setup
<..>

Cloning into '/home/wouter/workspace/rpi/yoe-distro/sources/meta-rust'...
Cloning into '/home/wouter/workspace/rpi/yoe-distro/sources/meta-ti'...
Cloning into '/home/wouter/workspace/rpi/yoe-distro/sources/meta-variscite-fslc'...
Cloning into '/home/wouter/workspace/rpi/yoe-distro/sources/meta-yoe'...
Cloning into '/home/wouter/workspace/rpi/yoe-distro/sources/openembedded-core'...
Submodule path 'sources/bitbake': checked out 'afee3f594e1510051a0b18e430e92549caf72fa2'
Submodule path 'sources/meta-96boards': checked out '7cbbce2aaebb5de4956db1a1d16134144e855f1a'
Submodule path 'sources/meta-browser': checked out 'b5fa3f598236a58da6217852340f5f8dd01c9a71'
Submodule path 'sources/meta-freescale': checked out '9594fe7247fdcc4bc6a5080c966545cdd5d2230b'
Submodule path 'sources/meta-freescale-3rdparty': checked out '8b290b151518d58cf89ccbd7fab32df3171d3d0e'
Submodule path 'sources/meta-intel': checked out 'b639f225503953945c1ebbc9dfcac6a30294a948'
Submodule path 'sources/meta-odroid': checked out '88f090bfb3d47c57abacd96a216a8eb851c24023'
Submodule path 'sources/meta-openembedded': checked out '97f56b745096a9cc3156b38f190d8aca6f56ce37'
Submodule path 'sources/meta-qcom': checked out 'a38028357962542f80a3075b7ec2e089fcc5d321'
Submodule path 'sources/meta-qt5': checked out '5832bf2747a70257eca5c443fd487532d62c90b3'
Submodule path 'sources/meta-raspberrypi': checked out 'b589574d9e7d56989bd117e3b140e10fd5426ad6'
Submodule path 'sources/meta-riscv': checked out 'c84c026c9417c38fc4baa90e873522dcc648e572'
error: Server does not allow request for unadvertised object 805b9bdeca065450c41eea1f8143b7c2fb7d82cb
Fetched in submodule path 'sources/meta-rust', but it did not contain 805b9bdeca065450c41eea1f8143b7c2fb7d82cb. Direct fetching of that commit failed.
```

And yoe does not properly initialize after that. 
If I follow the submodule reference I can see the commit exists: https://github.com/YoeDistro/meta-rust/commit/805b9bdeca065450c41eea1f8143b7c2fb7d82cb. But it seems the redirect does not work for me? 

Using git version 2.17.0 on Ubuntu 16.04.

cbrake, 2018-11-12:
I see the same thing here. @kraj do you have some changes that need to be pushed to YoeDistro/meta-rust or are we missing something? An older snapshot of meta-rust contains 805b9bdeca065450c41eea1f8143b7c2fb7d82cb but a new snapshot does not, so I suspect this commit got deleted git force push or something.

kraj, 2018-11-12:
sent a pull request just now https://github.com/YoeDistro/yoe-distro/pull/53

kraj, 2018-11-13:
should be fixed now

closed 2018-11-13 by kraj