YoeDistro/yoe-distro #61: support DL_DIR outside yoe-distro build dir

cbrake, 2018-11-21:
If users redefine DL_DIR outside the build directory, is there any way to map this directory in docker.
One possibility is to parse the auto/site/local.conf files in envsetup.sh and look for DL_DIR. Ben has used something like this:

```
if [ -f conf/site.conf ]; then
DL_DIR=$(cat conf/site.conf | grep DL_DIR | awk '{print$3}' | tr -d '"')
fi
```
But there may be other ways.

openity-ben, 2018-11-21:
Just to further clarify, once I parsed DL_DIR from the conf file, I then added the following as a mount point to the docker run command:

-v $DL_DIR:$DL_DIR \

My thought behind this approach was to have a common/shared downloads folder to use across multiple projects and repos without having to re-download packages for each build and consume unnecessary disk space

kraj, 2018-11-21:
I think this approach looks ok.

cbrake, 2018-12-07:
@openity-ben can you review/test this and see if it works for you:
https://github.com/YoeDistro/yoe-distro/pull/79

closed 2018-12-18 by cbrake