YoeDistro/yoe-distro #70: envsetup.sh - lines 37+38 called with each execution

novalisek, 2018-12-01:
Hello,
everytime I call raspberrypi3-envsetup.sh (on Linuxmint 19, ie. when I want to continue bitbaking after reboot), then lines 37+38 
 sudo aptitude install expect pv -y
 expect -c 'spawn sudo dpkg-reconfigure -freadline dash; send "n\n"; interact;'
from envsetup.sh are always called

$readlink /bin/sh  is returning  "dash", aptitude is also installed, then it's called ...  should there be "!=" instead of "=" ?  or some other conditions?
Or should I call some other script to be able to continue bitbaking?

Thanks.

kraj, 2018-12-02:
This does not happen here. The idea is to gently ask the user to use bash as default shell. Since Yoe has not been tested with other shells. I think if the setup detects dash as default shell then it will always come up

kraj, 2018-12-03:
this sounds a good idea. Would you mind cooking up a patch ?

novalisek, 2018-12-03:
ok, but I am still not getting point of lines 27 and 39 - if current shell is "dash", why should it say that dash is not supported and exit?

kraj, 2018-12-03:
> ah, yes ... but will lines 49+50 be ever called if script exits at line 38 with same condition?
> 

technically yes they will not be called. We can remove them.

> And should the message be "bash shell is not supported" instead of "dash shell is not supported" ?

bash is supported, dash is not. So message is correct.

novalisek, 2018-12-03:
> 
> you could still be in dash as interactive shell, even though your /bin/sh is pointing to bash

> bash is supported, dash is not. So message is correct.

So if I am in dash interactive shell, the it must be supported, or am I still missing something :) 
If you want to disable dash (and use bash), then you shouldn't return from the script if you are not in dash, or do you? 

Or should we better disable condition at line 36, because if I am in dash, then I will never get to "dpkg-reconfigure" to change /bin/sh link? then exit after "dpkg-reconfigure" and then user should try to start the script again?

kraj, 2018-12-03:
as of now, we have not tested dash as either interactive or non-interactive shell. It returns from script if it detects dash as your interactive shell, or bin/sh is pointing to dash. Two different case. I think we can do better here, where we can provide help so user can take actions on own instead of script trying to run installer and making changes to defaults etc.

dash needs testing as /bin/sh. The envsetup script itself needs bash.

kraj, 2018-12-03:
@cbrake is thinking about redesigning it a bit.

novalisek, 2018-12-03:
I understand that it exits if it detects dash, but why does it write, that dash is not supported, if you are in dash right now ... (I was thinking it should write that "bash" is not supported)

Could this work? 
[envsetup.patch.txt](https://github.com/YoeDistro/yoe-distro/files/2641272/envsetup.patch.txt)

cbrake, 2018-12-03:
we require bash shell, and that /bin/sh link to bash. 

This may be confusing, but the following:

`sudo expect -c 'spawn dpkg-reconfigure -freadline dash; send "n\n"; interact;'`

configures /bin/sh to point to bash, not dash. The logic for this just happens to be in the dash package.

Thus, the error message "dash as default shell is not supported"

What do you think of simplifying all this to:
https://github.com/YoeDistro/yoe-distro/pull/74

novalisek, 2018-12-03:
looking good and easy to understand to me :+1:

closed 2018-12-03 by novalisek

cbrake, 2018-12-03:
made one more tweak to this: https://github.com/YoeDistro/yoe-distro/pull/75

@novalisek let me know if this breaks anything on your end.

novalisek, 2018-12-03:
@cbrake seems to be working fine, does nothing under bash,  exits under dash