YoeDistro/yoe-distro #146: Can I use Https to init submodule?

JavaCS3, 2019-03-26:
My company firewall doesn't allow us to connect to Github via SSH directly. How can I replace all SSH connection to HTTPS?

cbrake, 2019-03-26:
so looking at https://github.com/YoeDistro/yoe-distro/blob/master/.gitmodules, it appears we are using the git protocol for most of the submodules (appears there are 3 git protocols https://git-scm.com/book/en/v2/Git-on-the-Server-The-Protocols). I would support changing all the git URLs in .gitmodules to use the https instead of the git protocol, as that is likely most compatible with company firewalls.

Thoughts @kraj ?

kraj, 2019-03-26:
@cbrake yes we used git but I would prefer https myself too. Care for a patch ?
 :)

closed 2019-03-27 by cbrake