YoeDistro/yoe-distro #147: docker: Operation not permitted

JavaCS3, 2019-03-26:
I set DOCKER_REPO and compile it as below but met this issue 
```
bitbake core-image-base
Setting MACHINE=raspberrypi3-64
/yoe-demo/yoe-distro/conf/auto.conf has been updated
Setting DOCKER_REPO to yoedistro/yoe-build:stretch
NOTE: Started PRServer with DBfile: /yoe-demo/yoe-distro/cache/prserv.sqlite3, IP: 127.0.0.1, PORT: 38290, PID: 49
Loading cache: 100% |############################################################################################################################################| Time: 0:00:01
Loaded 3409 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-yoe-linux"
MACHINE              = "raspberrypi3-64"
DISTRO               = "yoe-glibc-sysvinit-wayland"
DISTRO_VERSION       = "2.6"
TUNE_FEATURES        = "aarch64 cortexa53 crc"
TARGET_FPU           = ""
meta-yoe             = "HEAD:8b6033825498851498fb7ed1c4ea37078f8b38fe"
meta-raspberrypi     = "HEAD:7af614b2ee362765f299faaa732cafc436aa0ae1"
meta-filesystems
meta-gnome
meta-initramfs
meta-multimedia
meta-networking
meta-oe
meta-python
meta-webserver       = "HEAD:61c73067cb84ba2d7bef2da40f116f972686a20f"
meta                 = "HEAD:3da578a79195a63fb6d4bbd68d8f1548defd205f"

Initialising tasks: 100% |#######################################################################################################################################| Time: 0:00:03
Sstate summary: Wanted 727 Found 0 Missed 727 Current 434 (0% match, 37% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: cdrtools-native-3.01a31+really3.01-r0 do_install: Function failed: do_install (log file is located at /yoe-demo/yoe-distro/build/tmp/work/x86_64-linux/cdrtools-native/3.01a31+really3.01-r0/temp/log.do_install.136)
ERROR: Logfile of failure stored in: /yoe-demo/yoe-distro/build/tmp/work/x86_64-linux/cdrtools-native/3.01a31+really3.01-r0/temp/log.do_install.136
Log data follows:
| DEBUG: Executing shell function do_install
| 	==> MAKING "install" ON SUBDIRECTORY "SRCROOT/conf"
| make[1]: Entering directory '/yoe-demo/yoe-distro/build/tmp/work/x86_64-linux/cdrtools-native/3.01a31+really3.01-r0/cdrtools-3.01/conf'
| 	==> MAKING DIRECTORY "/yoe-demo/yoe-distro/build/tmp/work/x86_64-linux/cdrtools-native/3.01a31+really3.01-r0/image/yoe-demo/yoe-distro/build/tmp/work/x86_64-linux/cdrtools-native/3.01a31+really3.01-r0/recipe-sysroot-native/usr/include/schily/x86_64-linux-gcc"
| 	==> INSTALLING "/yoe-demo/yoe-distro/build/tmp/work/x86_64-linux/cdrtools-native/3.01a31+really3.01-r0/image/yoe-demo/yoe-distro/build/tmp/work/x86_64-linux/cdrtools-native/3.01a31+really3.01-r0/recipe-sysroot-native/usr/include/schily/x86_64-linux-gcc/xconfig.h"
| Intercept /yoe-demo/yoe-distro/sources/openembedded-core/scripts/native-intercept/chown: root /tmp/xd.302/xx.302 -- do nothing
| Intercept /yoe-demo/yoe-distro/sources/openembedded-core/scripts/native-intercept/chown: root /yoe-demo/yoe-distro/build/tmp/work/x86_64-linux/cdrtools-native/3.01a31+really3.01-r0/image/yoe-demo/yoe-distro/build/tmp/work/x86_64-linux/cdrtools-native/3.01a31+really3.01-r0/recipe-sysroot-native/usr/include/schily/x86_64-linux-gcc/#inst.302# -- do nothing
| chgrp: changing group of '/yoe-demo/yoe-distro/build/tmp/work/x86_64-linux/cdrtools-native/3.01a31+really3.01-r0/image/yoe-demo/yoe-distro/build/tmp/work/x86_64-linux/cdrtools-native/3.01a31+really3.01-r0/recipe-sysroot-native/usr/include/schily/x86_64-linux-gcc/#inst.302#': Operation not permitted
| ../RULES/rules.ins:29: recipe for target '/yoe-demo/yoe-distro/build/tmp/work/x86_64-linux/cdrtools-native/3.01a31+really3.01-r0/image/yoe-demo/yoe-distro/build/tmp/work/x86_64-linux/cdrtools-native/3.01a31+really3.01-r0/recipe-sysroot-native/usr/include/schily/x86_64-linux-gcc/xconfig.h' failed
| make[1]: *** [/yoe-demo/yoe-distro/build/tmp/work/x86_64-linux/cdrtools-native/3.01a31+really3.01-r0/image/yoe-demo/yoe-distro/build/tmp/work/x86_64-linux/cdrtools-native/3.01a31+really3.01-r0/recipe-sysroot-native/usr/include/schily/x86_64-linux-gcc/xconfig.h] Error 1
| make[1]: Leaving directory '/yoe-demo/yoe-distro/build/tmp/work/x86_64-linux/cdrtools-native/3.01a31+really3.01-r0/cdrtools-3.01/conf'
| RULES/rules1.dir:25: recipe for target 'install' failed
| make: *** [install] Error 2
| WARNING: /yoe-demo/yoe-distro/build/tmp/work/x86_64-linux/cdrtools-native/3.01a31+really3.01-r0/temp/run.do_install.136:1 exit 2 from 'make install GMAKE_NOWARN=true INS_BASE=/yoe-demo/yoe-distro/build/tmp/work/x86_64-linux/cdrtools-native/3.01a31+really3.01-r0/recipe-sysroot-native/usr DESTDIR=/yoe-demo/yoe-distro/build/tmp/work/x86_64-linux/cdrtools-native/3.01a31+really3.01-r0/image'
| ERROR: Function failed: do_install (log file is located at /yoe-demo/yoe-distro/build/tmp/work/x86_64-linux/cdrtools-native/3.01a31+really3.01-r0/temp/log.do_install.136)
ERROR: Task (/yoe-demo/yoe-distro/sources/openembedded-core/meta/recipes-devtools/cdrtools/cdrtools-native_3.01.bb:do_install) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1651 tasks of which 1636 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Build completion summary:
NOTE:   do_packagedata: 0.0% sstate reuse(0 setscene, 6 scratch)
NOTE:   do_package_write_ipk: 0.0% sstate reuse(0 setscene, 2 scratch)

Summary: 1 task failed:
  /yoe-demo/yoe-distro/sources/openembedded-core/meta/recipes-devtools/cdrtools/cdrtools-native_3.01.bb:do_install
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

cbrake, 2019-03-26:
can you try to run a chgrp command manually in docker?

```
dkr
chgrp ....
```

One thing to note -- the docker wrapper only really works well if your user ID is 1000. If your UID is not 1000, I recommend creating another user with that ID and running the builds as that user. (I need to update the documentation on this)

JavaCS3, 2019-03-26:
Why is that works well on UID=1000?
Seems that I need to mount /etc/passwd in docker or `docker run --privilege ...`
```
******@******:/yoe-demo/yoe-distro$ dkr
setting dkr action to shell
reading local settings
Setting MACHINE=raspberrypi3
/yoe-demo/yoe-distro/conf/auto.conf has been updated
I have no name!@543b54811e27:/yoe-demo/yoe-distro$ ls -al README.md
-rw-rw-r-- 1 1001 1001 6314 Mar 25 18:48 README.md
I have no name!@543b54811e27:/yoe-demo/yoe-distro$ chgrp root README.md
I have no name!@543b54811e27:/yoe-demo/yoe-distro$ ls -al README.md
-rw-rw-r-- 1 1001 root 6314 Mar 25 18:48 README.md
I have no name!@543b54811e27:/yoe-demo/yoe-distro$ chgrp 1001 README.md
chgrp: changing group of 'README.md': Operation not permitted
I have no name!@543b54811e27:/yoe-demo/yoe-distro$
```

cbrake, 2019-03-26:
The docker container has a user named "build" which is UID 1000. OE builds need to be run not as root, so we have not found a good way around this build/UID 1000 issue.

JavaCS3, 2019-03-27:
@cbrake I found the root cause. It seems that my host shell doesn't has $GID So it enter docker with $UID:root. I fixed it by manually export GID=`id -g`. I think it would be better to fix https://github.com/YoeDistro/yoe-distro/blob/master/envsetup.sh#L474
by `--user $(id -u):$(id -g)`

cbrake, 2019-03-27:
I think this is a good change -- do you want to send a pull request?

JavaCS3, 2019-03-27:
Sure, I'm new to contribute Open Source Project. Maybe you can help me.

closed 2019-03-29 by JavaCS3