YoeDistro/yoe-distro #221: SDK build issue

cbrake, 2019-09-25:
Getting errors when building SDK for SAMA5D27

[sdk-error.log](https://github.com/YoeDistro/yoe-distro/files/3653565/sdk-error.log)

kraj, 2019-09-25:
@cbrake is this using master ?

cbrake, 2019-09-25:
yes

kraj, 2019-09-25:
when `meta-clang` is included in layer mix then the default is to add clang to SDK, the error you are seeing is due to the fact that its trying to use gcc to compile libc++ ( clang runtime ) I don't know why thats happening, since libcxx recipe explicitly demands clang see [1]. In anycase it will be good for meta-clang to not insert itself unless someone wants clang in their or at least provide a knob for user to easily exclude it from SDK. See [2]

[1] https://github.com/kraj/meta-clang/blob/master/recipes-devtools/clang/libcxx_git.bb#L93
[2] https://github.com/kraj/meta-clang/pull/189

kraj, 2020-03-01:
`bitbake -cpopulate_sdk_ext yoe-simple-image` works ok for `sama5d27-som1-ek-sd` now on master

closed 2020-03-01 by kraj