YoeDistro/yoe-distro #248: raspberrypi3-64: "Failed to create a file with a long name."

kdsch, 2020-01-09:
```
s ▷ uname -a
Linux XPS-13-9360 4.15.0-72-generic #81-Ubuntu SMP Tue Nov 26 12:20:02 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
s ▷ git clone -q --recurse-submodules -j8 -b master git://github.com/YoeDistro/
yoe-distro.git yoe
s ▷ cd yoe
yoe ▷ . ./raspberrypi3-64-envsetup.sh
Setting MACHINE=raspberrypi3-64
YOE_ENV_VERSION mismatch, recreating localconfig.sh
localconfig.sh created
/home/karl/s/yoe/conf/auto.conf has been updated
yoe ▷ yoe_setup
yoe ▷ bitbake yoe-simple-image
Setting MACHINE=raspberrypi3-64
/home/karl/s/yoe/conf/auto.conf has been updated
Setting DOCKER_REPO to yoedistro/yoe-build:buster
NOTE: Started PRServer with DBfile: /home/karl/s/yoe/cache/prserv.sqlite3, IP:
127.0.0.1, PORT: 36453, PID: 29
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    Failed to create a file with a long name in TMPDIR. Please use a filesystem that does not unreasonably limit filename length.
Failed to create a file with a long name in SSTATE_DIR. Please use a filesystem that does not unreasonably limit filename length.

Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

kraj, 2020-01-09:
@kdsch thanks for reporting this problem. I have not seen this issue, so lets triage it here a bit. 
What is your host distro, and what shell is you default shell ?

cbrake, 2020-01-09:
what is the path of your Yoe checkout (is it long), and what file system are you using?

I use ext4 here.

kdsch, 2020-01-09:
@cbrake @kraj I've answered your questions in the original post.

cbrake, 2020-01-09:
a few things you could try: 

* try bash instead of dash -- seems like OE should be running in docker container, so seems this should not matter, unless the env is not getting set up right before docker runs.
* and source env setup, run `dkr`, and make sure `/etc/os-release` looks like:

```
build@386bd39209db:/scratch/oe/yoe-distro$ more /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

This will verify docker is somewhat working.

kdsch, 2020-01-09:
@cbrake I was wrong about the filesystem and the shell. I edited the post again.

kdsch, 2020-01-09:
https://unix.stackexchange.com/a/32834

kdsch, 2020-01-09:
The problem is ecryptfs.
https://software.intel.com/en-us/node/593591

closed 2020-01-09 by kdsch

kraj, 2020-01-09:
@kdsch yeah encryption takes a bit of max len limits for file and cmdline, so it reduces from 255 to something like 140 chars and recommendation is like 40 which could be conservative for file names.