YoeDistro/yoe-distro #305: drop meta-python2 from layermix

kraj, 2020-03-30:
python2 is EOL since Jan 2020, and so far we have had issues building other pieces with python3 but it seems at this point we can build pretty much everything in world builds barring few recipes.

cbrake, 2020-03-30:
sounds good to me

kraj, 2020-04-01:
hmm it seems meta-browser still needs it
```
ERROR: ParseError at /mnt/b/yoe/sources/meta-browser/dynamic-layers/rust-layer/recipes-browser/mozilla-devscripts/mozilla-devscripts_git.bb:15: Could not inherit file classes/distutils.bbclass

```

I wonder if we should just drop meta-browser too and replace it with meta-webkit

cbrake, 2020-04-01:
that seems fine to me -- the target applications for Yoe are embedded systems, so as long as we have one good browser solution, I don't think we need meta-browser.

commit 0eb56302 (2020-03-30): Delete meta-python2

Python2 is EOL, so lets remove it for upcoming 3.1 release

Fixes #305
closed 2020-04-01 by cbrake