YoeDistro/yoe-distro #356: use TLS capable wget in released images

cbrake, 2020-05-01:
avoid messages like:

 Downloading https://s3.wasabisys.com/yoe-feed/3.1/glibc/armv7at2hf-neon/go_1.14.1-r0.0_armv7at2hf-neon.ipk.

cbrake, 2020-05-01:
after running `opkg install wget`, the TLS messages go away.

It installs:

```
root@beaglebone:~# opkg install wget
Installing openssl-conf (1.1.1f) on root
Downloading https://s3.wasabisys.com/yoe-feed/3.1/glibc/armv7at2hf-neon/openssl-conf_1.1.1f-r0.0_armv7at2hf-neon.ipk.
wget: note: TLS certificate validation not implemented
Installing libcrypto1.1 (1.1.1f) on root
Downloading https://s3.wasabisys.com/yoe-feed/3.1/glibc/armv7at2hf-neon/libcrypto1.1_1.1.1f-r0.0_armv7at2hf-neon.ipk.
wget: note: TLS certificate validation not implemented
Installing libssl1.1 (1.1.1f) on root
Downloading https://s3.wasabisys.com/yoe-feed/3.1/glibc/armv7at2hf-neon/libssl1.1_1.1.1f-r0.0_armv7at2hf-neon.ipk.
wget: note: TLS certificate validation not implemented
Installing openssl-bin (1.1.1f) on root
Downloading https://s3.wasabisys.com/yoe-feed/3.1/glibc/armv7at2hf-neon/openssl-bin_1.1.1f-r0.0_armv7at2hf-neon.ipk.
wget: note: TLS certificate validation not implemented
Installing ca-certificates (20190110) on root
Downloading https://s3.wasabisys.com/yoe-feed/3.1/glibc/all/ca-certificates_20190110-r0.0_all.ipk.
wget: note: TLS certificate validation not implemented
Installing wget (1.20.3) on root
Downloading https://s3.wasabisys.com/yoe-feed/3.1/glibc/armv7at2hf-neon/wget_1.20.3-r0.0_armv7at2hf-neon.ipk.
wget: note: TLS certificate validation not implemented
Configuring openssl-conf.
Configuring libcrypto1.1.
Configuring libssl1.1.
Configuring openssl-bin.
Configuring ca-certificates.
Updating certificates in /etc/ssl/certs...
128 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
Configuring wget.
update-alternatives: Linking /usr/bin/wget to /usr/bin/wget.wget
```

kraj, 2020-05-01:
I think its busybox wget applet problem. This is default in yoe-simple-image and when you install GNU wget then update-alternative chose that to provide wget and hence problem disappears. But doing 

```
busybox wget <some URI>

```
should still have the issue

cbrake, 2020-05-01:
could we modify the released images to include wget to avoid bad impressions on first opkg isntall?

You know how sensitive people are to security these days :-)

kraj, 2020-05-01:
I think thats a fair point we should include wget in IMAGE_INSTALL for yoe-simple-image 
will do that for rc.3

kraj, 2020-05-01:
[Here](https://git.busybox.net/busybox/commit/?id=0972c7f7a570c38edb68e1c60a45614b7a7c7d55) is relevant patch in busybox that emits the error

closed 2020-05-04 by cbrake