YoeDistro/yoe-distro #376: iso installer for bare metal

outbackdingo, 2020-05-15:
more a question, i know there exists meta-anaconda which does a wic based iso install to bare metal, is there anything cpommparable in this yoe-distro that will generate an iso from an image and allow it to be installed to  bare metal ?? or are we are mostly dd'ing images to sdcards so far? which wount work on real hard drives for say x86_64 installation

cbrake, 2020-05-15:
I don't see any reason you could not include the meta-anaconda layer in your build. I've never tried it (mostly focused on ARM, etc), but looks like an interesting way to do an install in x86.

Might also be possible to expand one of the update mechanisms to do something similar. One x86 project we did we had a process to make a bootable USB disk that had an updater which then programmed the main system flash. However, this was fairly specific code to a project and we never made it generic enough to be generally useful.

outbackdingo, 2020-05-15:
The issue with integrating meta-anaconda is it requires a target build for the install image, and a install build for the iso image, which includes the target image. so not quite sure how id go about that under yoe. id probably have to create separate "environments"

kraj, 2020-05-16:
After you setup yoe env ( . ./qemux86-envsetup.sh) you would add meta-anacoda to setup

```
yoe_add_layer git://git.yoctoproject.org/meta-anaconda master
```

it needs systemd so use right Yoe profile edit `conf/site.conf`
```
YOE_PROFILE ?= "yoe-glibc-systemd-wayland"
```
then it requires some settings in local.conf
```
           $ echo 'DISTRO_FEATURES_append = " anaconda-support"' >> conf/local.conf
           $ echo 'PACKAGE_CLASSES = "package_rpm"' >> conf/local.conf' >> conf/local.conf

```

build target image

```
bitbake core-image-minimal
```
Now build installer image
```
echo 'INSTALLER_TARGET_BUILD = "<target_build>/tmp/deploy/images/qemux86/core-image-minimal-qemux86.ext4"' >> conf/local.conf
bitbake core-image-anaconda
```

You should then get ISO which can be burned to disk and installed

outbackdingo, 2020-05-16:
looks good only thing i question is the profile... its afor a i7 class x86_64 server vm.... no wayland needed, so guess i can copy the YOE_PROFILE ?= "yoe-glibc-systemd-wayland" and make a YOE_PROFILE ?= "yoe-glibc-systemd" one....

kraj, 2020-05-16:
No worries, wayland is only pulled in when you build an image with it. it just says which graphics and compositor will be used.