YoeDistro/yoe-distro #379: intel-corei7-64-envsetup.sh  - no meta-intel in bblayers

outbackdingo, 2020-05-16:
. ./intel-corei7-64-envsetup.sh 
Setting MACHINE=intel-corei7-64
YOE_ENV_VERSION mismatch, recreating localconfig.sh
localconfig.sh created
/home/dingo/yoe/conf/auto.conf has been updated
Setting DOCKER_REPO to yoedistro/yoe-build:buster

ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    MACHINE=intel-corei7-64 is invalid. Please set a valid MACHINE in your local.conf, environment or other configuration file.
BSPLAYERS = "\
  ${TOPDIR}/sources/meta-96boards \
  ${TOPDIR}/sources/meta-altera \
  ${TOPDIR}/sources/meta-arm/meta-arm \
  ${TOPDIR}/sources/meta-atmel \
  ${TOPDIR}/sources/meta-freescale \
  ${TOPDIR}/sources/meta-freescale-3rdparty \
  ${TOPDIR}/sources/meta-odroid \
  ${TOPDIR}/sources/meta-odroid/meta-odroid-extras \
  ${TOPDIR}/sources/meta-qcom \
  ${TOPDIR}/sources/meta-raspberrypi \
  ${TOPDIR}/sources/meta-riscv \
  ${TOPDIR}/sources/meta-ti \

no meta-intel in bblayers

kraj, 2020-05-16:
yes indeed, as @cbrake said we never had a production usecase for x86 besides minnowboards once. Perhaps you can add meta-intel via yoe_add_layer and also send a pull if you wish.

cbrake, 2020-05-16:
It is pretty standard to add meta layers as needed with OE. yoe_add_layer is just a bash function that automates and documents how to do it. You can read it in envsetup.sh if you want to understand how it works. meta-xyz are just git submodules.

kraj, 2022-05-02:
we do have meta-intel in .gitmodules. but not part of `conf/bblayers.conf` so just adding

```

  ${TOPDIR}/sources/meta-atmel \
```

to `BSPLAYERS` variable should get you going. 

Should we include meta-intel by default ? perhaps but we have no products currently needing it. So if someone steps up to support it we are good to add it.