YoeDistro/yoe-distro #410: dunfell C4 build issue

cbrake, 2020-06-18:
@kraj do you have any ideas of things to try on this one?

```
/yoe/yoe-distro-dunfell/build/tmp/work/aarch64-yoe-linux/gcc-runtime/9.3.0-r0/gcc-9.3.0/build.aarch64-yoe-linux.aarch64-yoe-linux=/usr/src/debug/gcc-runtime/9.3.0-r0 -pthread -DN=2 -DIFUNC_ALT=1 -march=armv8-a+lse -c ../../../../../../../../work-shared/gcc-9.3.0-r0/gcc-9.3.0/libatomic/load_n.c  -fPIC -DPIC -o .libs/load_2_1_.o
| cc1: all warnings being treated as errors
| cc1: error: switch '-mcpu=armv8.2-a' conflicts with '-march=armv8-a' switch [-Werror]
| make[2]: *** [Makefile:855: fand_1_1_.lo] Error 1
| cc1: error: switch '-mcpu=armv8.2-a' conflicts with '-march=armv8-a' switch [-Werror]
| cc1: error: switch '-mcpu=armv8.2-a' conflicts with '-march=armv8-a' switch [-Werror]
| cc1: error: switch '-mcpu=armv8.2-a' conflicts with '-march=armv8-a' switch [-Werror]
| cc1: all warnings being treated as errors
| make[2]: *** [Makefile:855: exch_1_1_.lo] Error 1
| cc1: all warnings being treated as errors
| make[2]: *** [Makefile:855: fior_1_1_.lo] Error 1
| cc1: all warnings being treated as errors
| cc1: all warnings being treated as errors
| make[2]: *** [Makefile:855: tas_1_1_.lo] Error 1
| make[2]: *** [Makefile:855: fxor_1_1_.lo] Error 1
| cc1: all warnings being treated as errors
| cc1: all warnings being treated as errors
| make[2]: *** [Makefile:855: load_2_1_.lo] Error 1
| make[2]: *** [Makefile:855: fnand_1_1_.lo] Error 1
| make[2]: Leaving directory '/scratch/yoe/yoe-distro-dunfell/build/tmp/work/aarch64-yoe-linux/gcc-runtime/9.3.0-r0/gcc-9.3.0/build.aarch64-yoe-linux.aarch64-yoe-linux/aarch64-yoe-linux/libatomic'
| make[1]: *** [Makefile:606: all-recursive] Error 1
| make[1]: Leaving directory '/scratch/yoe/yoe-distro-dunfell/build/tmp/work/aarch64-yoe-linux/gcc-runtime/9.3.0-r0/gcc-9.3.0/build.aarch64-yoe-linux.aarch64-yoe-linux/aarch64-yoe-linux/libatomic'
| make: *** [Makefile:446: all] Error 2
| WARNING: /scratch/yoe/yoe-distro-dunfell/build/tmp/work/aarch64-yoe-linux/gcc-runtime/9.3.0-r0/temp/run.do_compile.1039531:1 exit 1 from 'exit 1'
| 
ERROR: Task (/scratch/yoe/yoe-distro-dunfell/sources/openembedded-core/meta/recipes-devtools/gcc/gcc-runtime_9.3.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1972 tasks of which 2 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 2 seconds
NOTE: Build completion summary:
NOTE:   do_populate_sysroot: 0.0% sstate reuse(0 setscene, 132 scratch)
NOTE:   do_package_qa: 0.0% sstate reuse(0 setscene, 16 scratch)
NOTE:   do_package: 0.0% sstate reuse(0 setscene, 22 scratch)
NOTE:   do_packagedata: 0.0% sstate reuse(0 setscene, 21 scratch)
NOTE:   do_package_write_ipk: 0.0% sstate reuse(0 setscene, 16 scratch)
NOTE:   do_populate_lic: 0.0% sstate reuse(0 setscene, 267 scratch)

Summary: 1 task failed:
  /scratch/yoe/yoe-distro-dunfell/sources/openembedded-core/meta/recipes-devtools/gcc/gcc-runtime_9.3.bb:do_compile
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
[cbrake@ceres yoe-distro-dunfell]$ pwd
/scratch/yoe/yoe-distro-dunfell
```

kraj, 2020-06-18:
are you on latest dunfell ? whats your oe-core top rev ?

kraj, 2020-06-18:
Should be fixed by https://github.com/YoeDistro/yoe-distro/pull/411

cbrake, 2020-06-19:
looks like it is fixed now -- thanks!

closed 2020-06-19 by cbrake