YoeDistro/yoe-distro #429: working with podman

outbackdingo, 2020-07-06:
Hi, development machine is now Fedora Silverblue 32, it utilizes podman ... docker like but not docker, any posibilities to support podman along with docker?

cbrake, 2020-07-06:
interesting idea.

You could try substituting podman for docker in this function:

https://github.com/YoeDistro/yoe-distro/blob/dunfell/envsetup.sh#L466

Maybe podman would handle the issue where UID is not 1000 better than docker (https://github.com/YoeDistro/yoe-distro/issues/7)

philn, 2020-12-03:
Additionally the volumes need to be mounted with the `:z` or `:Z` option, otherwise the envsetup.sh eval within the sandbox will fail. But I still get errors about some folders being read-only... need to debug this further

cbrake, 2020-12-07:
@philn thanks for your input on this -- I've not had time to try this, but podman looks very interesting for this application.

commit 9c46104b (2020-12-15): Add podman support

By default docker is used if DOCKER_REPO is set to a value different than
none. If you prefer to use podman instead of docker, un-comment this line

This was tested on a Silverblue F33 host.

Fixes #429
closed 2020-12-15 by kraj