YoeDistro/yoe-distro #468: e-sdk does not build

karthikpoduval, 2020-09-12:
Trying to build esdk for qemuarm64.

```
bitbake core-image-minimal -c do_populate_sdk_ext
```
However it fails at the end looking for local.conf which isn't present.

```
ERROR: core-image-minimal-1.0-r0 do_populate_sdk_ext: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:copy_buildsystem(d)
     0003:
File: '/media/kpoduval/hdd31/yoe-workspace/sources/openembedded-core/meta/classes/populate_sdk_ext.bbclass', lineno: 313, function: copy_buildsystem
     0309:                oldlines += f.readlines()
     0310:        if os.path.exists(builddir + '/conf/auto.conf'):
     0311:            with open(builddir + '/conf/auto.conf', 'r') as f:
     0312:                oldlines += f.readlines()
 *** 0313:        with open(builddir + '/conf/local.conf', 'r') as f:
     0314:            oldlines += f.readlines()
     0315:        (updated, newlines) = bb.utils.edit_metadata(oldlines, varlist, handle_var)
     0316:
     0317:        with open(baseoutpath + '/conf/local.conf', 'w') as f:
Exception: FileNotFoundError: [Errno 2] No such file or directory: '/media/kpoduval/hdd31/yoe-workspace/conf/local.conf'

ERROR: Logfile of failure stored in: /media/kpoduval/hdd31/yoe-workspace/build/tmp/work/qemuarm64-yoe-linux/core-image-minimal/1.0-r0/temp/log.do_populate_sdk_ext.31062
ERROR: Task (/media/kpoduval/hdd31/yoe-workspace/sources/openembedded-core/meta/recipes-core/images/core-image-minimal.bb:do_populate_sdk_ext) failed with exit code '1'

```

I had to create a local.conf with contents of conf/site.conf and MACHINE variable. Is there a better way ?

kraj, 2020-09-12:
thanks for report @karthikpoduval I have seen this error too in past so I think there is some room for improvement, I think by default Yoe does not create a `local.conf` and perhaps thats whats happening here. Can you check if you have a `conf/local.conf` in your build dir ?

If creating an empty local.conf is what fixes this issue then perhaps its an issue with OE-Core and we should open a bug
with yocto project bugzilla as well so  it can be fixed there to handle absense of local.conf graciously.

kraj, 2020-09-13:
@karthikpoduval  a fix has been posted here https://patchwork.openembedded.org/patch/176454/ please test it out in your environment and report back if it fixes the issue

karthikpoduval, 2020-09-15:
@kraj thanks for the patch, it fixes the issue and build succeeded.