YoeDistro/yoe-distro #481: bitbake yoe-simple-image - FAILED in zsh

aj700, 2020-10-02:
FYI.
for me "bitbake yoe-simple-image" step failed immediately
`docker: Error response from daemon: create  /home/<user>/workspace/yoe/downloads: " /home/<user>/workspace/yoe/downloads" includes invalid characters for a local volume name, only "[a-zA-Z0-9][a-zA-Z0-9_.-]" are allowed. If you intended to pass a host directory, use absolute path.`

my default shell is zsh. 
switched to "bash" and bitbake started action.

kraj, 2020-10-03:
@aj700 thanks for reporting the issue and using yoe distro !
I am a zsh user myself but have not encountered this error. I will try to reproduce this here as well, meanwhile it you have more info to add to reproduce it, that will help. I think its the presence of ':' in pathname which is tripping the mkdir but how does this get appended is the question.

kraj, 2020-10-08:
Hi @aj700 I tried to reproduce this issue so I am using latest archlinux and using zsh (version 5.8) and I followed https://github.com/YoeDistro/yoe-distro#workspace-setup

it worked ok for me. So I am not sure how to get to the issue you are seeing. can you tell more about your build
host, what filesystem is it running and so on ?

kraj, 2020-10-08:
@aj700 can you try adding  `set -x` at the top of `envsetup.sh` and then run the setup and I am interested in seeing the `docker run ...` command

Justkant, 2020-11-25:
Hi @kraj, I stumbled upon the same issue, here is the `dkr` command:
```zsh
+dkr:39> docker run --rm -i -t -v /home/kant/Documents/Projects/yoe:/home/kant/Documents/Projects/yoe -v /home/kant/.ssh:/home/build/.ssh '-v /home/kant/.gitconfig:/home/build/.gitconfig' '-v /home/kant/Documents/Projects/yoe/downloads:/home/kant/Documents/Projects/yoe/downloads' '-v /home/kant/Documents/Projects/yoe/build/sstate-cache:/home/kant/Documents/Projects/yoe/build/sstate-cache' '-v /home/kant/Documents/Projects/yoe/build/tmp:/home/kant/Documents/Projects/yoe/build/tmp' -v /tmp/ssh-kQMqY6uJlwIa/agent.4050:/ssh-agent -e 'SSH_AUTH_SOCK=/ssh-agent' -e 'MACHINE=tinker-board' -w /home/kant/Documents/Projects/yoe --user 1000:985 yoedistro/yoe-build:buster /bin/bash -c '. /home/kant/Documents/Projects/yoe/envsetup.sh && /home/kant/Documents/Projects/yoe/sources/bitbake/bin/bitbake yoe-simple-image '
docker: Error response from daemon: create  /home/kant/.gitconfig: " /home/kant/.gitconfig" includes invalid characters for a local volume name, only "[a-zA-Z0-9][a-zA-Z0-9_.-]" are allowed. If you intended to pass a host directory, use absolute path.

```

I'm also using zsh 5.8 with zgen on Arch Linux 5.9.9-arch1-1

PR #505 (2020-11-25): fix: bitbake yoe-simple-image using zsh
fixes #481
commit 646331b8 (2020-11-25): fix: bitbake yoe-simple-image using zsh

fixes #481
commit 2928816f (2020-11-26): fix: bitbake yoe-simple-image using zsh

fixes #481
closed 2020-11-25 by cbrake