YoeDistro/yoe-distro #494: rebuild docker wrapper using Ubuntu

cbrake, 2020-11-12:
Khem: python3 on ubuntu is LTO optimized and is fastest

cbrake, 2020-11-12:
started dockerfile here: https://github.com/YoeDistro/docker-yoe-build/tree/ubuntu_20.04

cbrake, 2020-11-12:
dealing with:

```
Could not configure 'libc6:i386'. 
 Could not perform immediate configuration on 'libgcc-s1:i386'. Please see man 5 apt.conf under APT::Immediate-Configure for details. (2)
```

kraj, 2020-11-12:
can we try 
```
sudo apt-get install -y libc6-i386
```

cbrake, 2020-11-12:
that works fine. I can't retrigger the problem again in the docker instance, even when I remove and re-install libgcc-s1:i386.

kraj, 2020-11-12:
ok cool. Lets use libc6-i386 as it seems to be more standard way

cbrake, 2020-11-12:
same thing:

```
Running hooks in /etc/ca-certificates/update.d...
done.
E: Could not configure 'libc6:i386'. 
E: Could not perform immediate configuration on 'libgcc-s1:i386'. Please see man 5 apt.conf under APT::Immediate-Configure for details. (2)
The command '/bin/sh -c dpkg --add-architecture i386 &&         apt-get update &&      apt-get install -yq sudo build-essential git      python python3 man bash diffstat gawk chrpath wget cpio        texinfo lzop apt-utils bc screen libncurses5-dev locales           libc6-i386 doxygen libssl-dev dos2unix xvfb x11-utils      g++-multilib libssl-dev:i386 zlib1g-dev:i386    libtool libtool-bin procps python3-distutils pigz socat &&   rm -rf /var/lib/apt-lists/* &&  echo "dash dash/sh boolean false" | debconf-set-selections &&  dpkg-reconfigure dash' returned a non-zero code: 100

```

kraj, 2020-11-12:
I wonder if we need `/bin/sh -c dpkg --add-architecture i386`

kraj, 2020-11-12:
I wonder if we are experiencing https://bugs.launchpad.net/ubuntu/+source/glibc/+bug/1871268

kraj, 2022-05-02:
We do have ubuntu-2020 docker image available now a days. 
https://hub.docker.com/layers/128720159/yoedistro/yoe-build/ubuntu_20.04/images/sha256-667fe0b61d2b48fce1a74f1b54b62ed8e540dd005649f17c70d5dd8ceb023708?context=repo

closed 2022-05-02 by kraj