YoeDistro/yoe-distro #555: raspberrypi0-wifi meta-clang compiler-rt arm/sync_fetch_and_add assembler requires armv8 and thumb2

mr-nice, 2021-06-12:
Hi all,
I'm not sure if this is the right place to report that bug or if I have to place it directly at meta-clang. But as this has to do with pulling in clang compiler-rt for raspberrypi0-wifi configuration I report it here. The problem is that compiler-rt tries to compile some assembly which requires armv8 and thumb2 instruction set. Which the raspberrypi0 is not capable of.

I didn't find a way to tell compiler-rt to not use the sync_fetch_and_add assembler files. So the solution might be to remove the dependency on compiler-rt for that target?

Build Configuration:
BB_VERSION           = "1.51.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-yoe-linux-gnueabi"
MACHINE              = "raspberrypi0-wifi"
DISTRO               = "yoe"
DISTRO_VERSION       = "2021.06-rc.1"
TUNE_FEATURES        = "arm thumb vfp arm1176jzfs callconvention-hard"
TARGET_FPU           = "hard"
meta-yoe             = "master:f36a2e949b42a28a746f75d09895635bdd7f325b"
meta-96boards        = "HEAD:533b7b3bb2dd883f03da79c3d91b622204b81cef"
meta-arm             
meta-arm-toolchain   = "HEAD:40611d12f5afaa7672659175f38b1e4cfd969283"
meta-atmel           = "HEAD:23e155839c59d0778c0333086a4dac6fe06cae07"
meta-freescale       = "HEAD:fbcff7045a9afbb383f52c2b6161dc2ce37be0e2"
meta-freescale-3rdparty = "HEAD:5f6c423aae0bc614d70e0d7b9bcc78e90b3711bb"
meta-odroid          
meta-odroid-extras   = "HEAD:55fb32367547491a06e1f524125abb6496315ae1"
meta-qcom            = "HEAD:a238fd051b585c725a3fb6b1a37731b622efbb94"
meta-raspberrypi     = "HEAD:7057d5dcbfa5802a370aa1730e14e940b9bf7b24"
meta-riscv           = "HEAD:f00613c0677497a7bf4ea536f31eeb8cbc5f6c23"
meta-rockchip        = "HEAD:838514a031e30112a7d00f2e20bf7e41264912bc"
meta-ti              = "HEAD:d54da744941ff97f2502bad2d7c51c5387bde929"
meta-clang           = "HEAD:040ec88b95679a953bf63461f73700246ed34eef"
meta-elm-binary      = "HEAD:612d7b08ea064288d6297d10d4cf8c2fb3606379"
meta-qt5             = "HEAD:a2882fecc96019781880c38fc6a81a9b71c944ad"
meta-rust            = "HEAD:ab86e0a068e90c361b803f59a4baa79bf0c92ba7"
meta-filesystems     
meta-gnome           
meta-initramfs       
meta-multimedia      
meta-networking      
meta-oe              
meta-perl            
meta-python          
meta-webserver       
meta-xfce            = "HEAD:c062b919c459e614a37f89b64caf9f8f066076ba"
meta                 = "HEAD:5799a807eed44d380e142913e50308589b9412ed"

Initialising tasks: 100% |###########################################################################################################################################################################| Time: 0:00:02
Sstate summary: Wanted 209 Local 0 Network 0 Missed 209 Current 291 (0% match, 58% complete)
NOTE: Executing Tasks
ERROR: compiler-rt-13.0.0-r0 do_compile: Execution of '/home/nice/develop/yoe/build/tmp/work/arm1176jzfshf-vfp-yoe-linux-gnueabi/compiler-rt/13.0.0-r0/temp/run.do_compile.131' failed with exit code 1:
[1/194] Building ASM object projects/compiler-rt/lib/builtins/CMakeFiles/clang_rt.builtins-armhf.dir/arm/sync_fetch_and_and_4.S.o
FAILED: projects/compiler-rt/lib/builtins/CMakeFiles/clang_rt.builtins-armhf.dir/arm/sync_fetch_and_and_4.S.o 
/home/nice/develop/yoe/build/tmp/work/arm1176jzfshf-vfp-yoe-linux-gnueabi/compiler-rt/13.0.0-r0/recipe-sysroot-native/usr/bin/arm-yoe-linux-gnueabi/arm-yoe-linux-gnueabi-clang -DVISIBILITY_HIDDEN -D_DEBUG -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE -D_LARGEFILE_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -Iprojects/compiler-rt/lib/builtins -I/home/nice/develop/yoe/build/tmp/work-shared/llvm-project-source-13.0.0-r0/git/compiler-rt/lib/builtins -Iinclude -I/home/nice/develop/yoe/build/tmp/work-shared/llvm-project-source-13.0.0-r0/git/llvm/include -target arm-yoe-linux-gnueabi  -marm -mfpu=vfp -mfloat-abi=hard -mcpu=arm1176jzf-s   -mlittle-endian -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/nice/develop/yoe/build/tmp/work/arm1176jzfshf-vfp-yoe-linux-gnueabi/compiler-rt/13.0.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/nice/develop/yoe/build/tmp/work/arm1176jzfshf-vfp-yoe-linux-gnueabi/compiler-rt/13.0.0-r0=/usr/src/debug/compiler-rt/13.0.0-r0                      -fdebug-prefix-map=/home/nice/develop/yoe/build/tmp/work/arm1176jzfshf-vfp-yoe-linux-gnueabi/compiler-rt/13.0.0-r0=/usr/src/debug/compiler-rt/13.0.0-r0                      -fdebug-prefix-map=/home/nice/develop/yoe/build/tmp/work/arm1176jzfshf-vfp-yoe-linux-gnueabi/compiler-rt/13.0.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/nice/develop/yoe/build/tmp/work/arm1176jzfshf-vfp-yoe-linux-gnueabi/compiler-rt/13.0.0-r0/recipe-sysroot-native=  -target arm-yoe-linux-gnueabi  -marm -mfpu=vfp -mfloat-abi=hard -mcpu=arm1176jzf-s   -mlittle-endian -Qunused-arguments -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/nice/develop/yoe/build/tmp/work/arm1176jzfshf-vfp-yoe-linux-gnueabi/compiler-rt/13.0.0-r0/recipe-sysroot -fPIC -g  -march=armv7-a -mfloat-abi=hard -fno-lto -DCOMPILER_RT_HAS_FLOAT16 -std=c11 -fomit-frame-pointer -DCOMPILER_RT_ARMHF_TARGET -MD -MT projects/compiler-rt/lib/builtins/CMakeFiles/clang_rt.builtins-armhf.dir/arm/sync_fetch_and_and_4.S.o -MF projects/compiler-rt/lib/builtins/CMakeFiles/clang_rt.builtins-armhf.dir/arm/sync_fetch_and_and_4.S.o.d -o projects/compiler-rt/lib/builtins/CMakeFiles/clang_rt.builtins-armhf.dir/arm/sync_fetch_and_and_4.S.o -c /home/nice/develop/yoe/build/tmp/work-shared/llvm-project-source-13.0.0-r0/git/compiler-rt/lib/builtins/arm/sync_fetch_and_and_4.S
/home/nice/develop/yoe/build/tmp/work-shared/llvm-project-source-13.0.0-r0/git/compiler-rt/lib/builtins/arm/sync_fetch_and_and_4.S:18:193: error: invalid instruction
.p2align 2; .thumb; .syntax unified; .arm ; ; .globl __sync_fetch_and_and_4 ; .type __sync_fetch_and_and_4,%function ; .hidden __sync_fetch_and_and_4 ; ; .thumb_func ; __sync_fetch_and_and_4: mcr p15, #0, r0, c7, c10, #5; mov r12, r0; .L_tryatomic_and_4 : ldrex r0, [r12]; and r2, r0, r1; strex r3, r2, [r12]; cmp r3, #0; bne .L_tryatomic_and_4; mcr p15, #0, r0, c7, c10, #5; bx lr
                                                                                                                                                                                                ^
/home/nice/develop/yoe/build/tmp/work-shared/llvm-project-source-13.0.0-r0/git/compiler-rt/lib/builtins/arm/sync_fetch_and_and_4.S:18:257: error: invalid instruction, any one of the following would fix this:
.p2align 2; .thumb; .syntax unified; .arm ; ; .globl __sync_fetch_and_and_4 ; .type __sync_fetch_and_and_4,%function ; .hidden __sync_fetch_and_and_4 ; ; .thumb_func ; __sync_fetch_and_and_4: mcr p15, #0, r0, c7, c10, #5; mov r12, r0; .L_tryatomic_and_4 : ldrex r0, [r12]; and r2, r0, r1; strex r3, r2, [r12]; cmp r3, #0; bne .L_tryatomic_and_4; mcr p15, #0, r0, c7, c10, #5; bx lr
                                                                                                                                                                                                                                                                ^
/home/nice/develop/yoe/build/tmp/work-shared/llvm-project-source-13.0.0-r0/git/compiler-rt/lib/builtins/arm/sync_fetch_and_and_4.S:18:257: note: instruction requires: arm-mode
.p2align 2; .thumb; .syntax unified; .arm ; ; .globl __sync_fetch_and_and_4 ; .type __sync_fetch_and_and_4,%function ; .hidden __sync_fetch_and_and_4 ; ; .thumb_func ; __sync_fetch_and_and_4: mcr p15, #0, r0, c7, c10, #5; mov r12, r0; .L_tryatomic_and_4 : ldrex r0, [r12]; and r2, r0, r1; strex r3, r2, [r12]; cmp r3, #0; bne .L_tryatomic_and_4; mcr p15, #0, r0, c7, c10, #5; bx lr
                                                                                                                                                                                                                                                                ^
/home/nice/develop/yoe/build/tmp/work-shared/llvm-project-source-13.0.0-r0/git/compiler-rt/lib/builtins/arm/sync_fetch_and_and_4.S:18:257: note: instruction requires: armv8m.base
[... cut ...]
	#4: main, /home/nice/develop/yoe/build/tmp/work/arm1176jzfshf-vfp-yoe-linux-gnueabi/compiler-rt/13.0.0-r0/temp/run.do_compile.131, line 171

Backtrace (metadata-relative locations):
	#1: cmake_runcmake_build, /home/nice/develop/yoe/sources/openembedded-core/meta/classes/cmake.bbclass, line 207
	#2: cmake_do_compile, /home/nice/develop/yoe/sources/openembedded-core/meta/classes/cmake.bbclass, line 211
	#3: do_compile, autogenerated, line 2
ERROR: Logfile of failure stored in: /home/nice/develop/yoe/build/tmp/work/arm1176jzfshf-vfp-yoe-linux-gnueabi/compiler-rt/13.0.0-r0/temp/log.do_compile.131

kraj, 2021-06-12:
@mr-nice thanks for the report. Its good to file issues here since we are now using clang compiler for yoe distro.  I will take a look and try to see whats going on

mr-nice, 2021-06-12:
Thanks! It is really nice to see how far clang has come :)

kraj, 2021-06-12:
Please try applying this patch in `work-shared/llvm-project-source-13.0.0-r0/git` and see if this helps

```patchdiff --git a/compiler-rt/lib/builtins/arm/sync-ops.h b/compiler-rt/lib/builtins/arm/sync-ops.h
index 7a26170741ad..d914f9d3a109 100644
--- a/compiler-rt/lib/builtins/arm/sync-ops.h
+++ b/compiler-rt/lib/builtins/arm/sync-ops.h
@@ -16,9 +16,8 @@

 #define SYNC_OP_4(op)                                                          \
   .p2align 2;                                                                  \
-  .thumb;                                                                      \
   .syntax unified;                                                             \
-  DEFINE_COMPILERRT_THUMB_FUNCTION(__sync_fetch_and_##op)                      \
+  DEFINE_COMPILERRT_FUNCTION(__sync_fetch_and_##op)                            \
   DMB;                                                                         \
   mov r12, r0;                                                                 \
   LOCAL_LABEL(tryatomic_##op) : ldrex r0, [r12];                               \
@@ -31,9 +30,8 @@

 #define SYNC_OP_8(op)                                                          \
   .p2align 2;                                                                  \
-  .thumb;                                                                      \
   .syntax unified;                                                             \
-  DEFINE_COMPILERRT_THUMB_FUNCTION(__sync_fetch_and_##op)                      \
+  DEFINE_COMPILERRT_FUNCTION(__sync_fetch_and_##op)                            \
   push {r4, r5, r6, lr};                                                       \
   DMB;                                                                         \
   mov r12, r0;                                                                 \
```

mr-nice, 2021-06-12:
Thanks! That worked for me :)
I will build the image and try to use it. As far as I see if something would not work with that patch one would see that fast :D

closed 2021-06-12 by mr-nice