YoeDistro/yoe-distro #560: nativesdk-clang: CMAKE_(C|CXX|ASM)_COMPILER is not a full path and was not found in the PATH.

mr-nice, 2021-06-24:
Hi all,
I want to build meta-toolchain as part of it nativesdk-clang is build. nativesdk-clang want's to use x86_64-yoesdk-linux-gcc which is not set approptiate and the do_configure step complains about it. I found the commit 7b687f1  which suggests to set TOOLCHAIN_class-nativesdk = "clang".  I added this to my local.conf without the effect of using clang as C or CXX compiler to build nativesdk-clang

As yoe is using clang as it's main compiler now I wonder what is the best approach to solve the problem? x86_64-yoesdk-linux-gcc is not added to the packages sysroot adding a dependency to it might solve that but a better approach might be to fix TOOLCHAIN_class-nativesdk = "clang" ? to use clang to compile it?

What is your opinion to that options? I can try to fix it and do a pull request but I wanted to know about the options first.

Thanks in advance!

ERROR: nativesdk-clang-13.0.0-r0 do_configure: Execution of '/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0/temp/run.do_configure.134' failed with exit code 1:
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: x86_64-yoesdk-linux-gcc
CMake Error at CMakeLists.txt:44 (project):
  The CMAKE_C_COMPILER:

    x86_64-yoesdk-linux-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:44 (project):
  The CMAKE_CXX_COMPILER:

    x86_64-yoesdk-linux-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:44 (project):
  The CMAKE_ASM_COMPILER:

    x86_64-yoesdk-linux-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0/build/CMakeFiles/CMakeOutput.log".
See also "/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0/build/CMakeFiles/CMakeError.log".
WARNING: /home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0/temp/run.do_configure.134:167 exit 1 from 'cmake -G 'Ninja' -DCMAKE_MAKE_PROGRAM=ninja $oecmake_sitefile /home/nice/develop/yoe/build/tmp/work-shared/llvm-project-source-13.0.0-r0/git/llvm -DCMAKE_INSTALL_PREFIX:PATH=/opt/yoe/2021.06-rc.1/sysroots/x86_64-yoesdk-linux/usr -DCMAKE_INSTALL_BINDIR:PATH=bin -DCMAKE_INSTALL_SBINDIR:PATH=bin -DCMAKE_INSTALL_LIBEXECDIR:PATH=libexec -DCMAKE_INSTALL_SYSCONFDIR:PATH=/opt/yoe/2021.06-rc.1/sysroots/x86_64-yoesdk-linux/etc -DCMAKE_INSTALL_SHAREDSTATEDIR:PATH=../com -DCMAKE_INSTALL_LOCALSTATEDIR:PATH=/opt/yoe/2021.06-rc.1/sysroots/x86_64-yoesdk-linux/var -DCMAKE_INSTALL_LIBDIR:PATH=lib -DCMAKE_INSTALL_INCLUDEDIR:PATH=include -DCMAKE_INSTALL_DATAROOTDIR:PATH=share -DPYTHON_EXECUTABLE:PATH=/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0/recipe-sysroot-native/usr/bin/python3-native/python3 -DPython_EXECUTABLE:PATH=/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0/recipe-sysroot-native/usr/bin/python3-native/python3 -DPython3_EXECUTABLE:PATH=/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0/recipe-sysroot-native/usr/bin/python3-native/python3 -DLIB_SUFFIX= -DCMAKE_INSTALL_SO_NO_EXE=0 -DCMAKE_TOOLCHAIN_FILE=/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0/toolchain.cmake -DCMAKE_NO_SYSTEM_FROM_IMPORTED=1 -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DLLVM_ENABLE_ASSERTIONS=OFF -DLLVM_ENABLE_EXPENSIVE_CHECKS=OFF -DLLVM_ENABLE_PIC=ON -DLLVM_BINDINGS_LIST='' -DLLVM_ENABLE_FFI=ON -DFFI_INCLUDE_DIR=$(pkg-config --variable=includedir libffi) -DLLVM_OPTIMIZED_TABLEGEN=ON -DLLVM_BUILD_EXTERNAL_COMPILER_RT=ON -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS_RELEASE=' -O2 -pipe -fmacro-prefix-map=/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0=/usr/src/debug/nativesdk-clang/13.0.0-r0                      -fdebug-prefix-map=/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0=/usr/src/debug/nativesdk-clang/13.0.0-r0                      -fdebug-prefix-map=/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0/recipe-sysroot-native=  -DNDEBUG -g0' -DCMAKE_C_FLAGS_RELEASE=' -O2 -pipe -fmacro-prefix-map=/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0=/usr/src/debug/nativesdk-clang/13.0.0-r0                      -fdebug-prefix-map=/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0=/usr/src/debug/nativesdk-clang/13.0.0-r0                      -fdebug-prefix-map=/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0/recipe-sysroot-native=  -DNDEBUG -g0' -DBUILD_SHARED_LIBS=OFF -DLLVM_ENABLE_PROJECTS='clang;clang-tools-extra;lld;lldb' -DLLVM_BINUTILS_INCDIR=/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0/recipe-sysroot/opt/yoe/2021.06-rc.1/sysroots/x86_64-yoesdk-linux/usr/include -DLLVM_TEMPORARILY_ALLOW_OLD_TOOLCHAIN=ON -DLLVM_TARGETS_TO_BUILD='AMDGPU;AArch64;ARM;BPF;Mips;PowerPC;RISCV;X86' -DLLVM_EXPERIMENTAL_TARGETS_TO_BUILD=';' -DCLANG_DEFAULT_RTLIB=compiler-rt -DLLVM_ENABLE_EH=ON -DCLANG_DEFAULT_CXX_STDLIB=libc++ -DLLVM_ENABLE_LIBEDIT=ON -DLLDB_ENABLE_LIBEDIT=ON -DCLANG_DEFAULT_OPENMP_RUNTIME=libomp -DLLDB_EDITLINE_USE_WCHAR=0 -DLLVM_ENABLE_LIBPFM=OFF -DLLVM_ENABLE_RTTI=ON -DLLVM_BUILD_LLVM_DYLIB=ON -DLLVM_LINK_LLVM_DYLIB=ON -DLLVM_USE_SPLIT_DWARF=OFF -DLLVM_ENABLE_TERMINFO=OFF -DCOMPILER_RT_TERMINFO_LIB=OFF -DCLANG_DEFAULT_UNWINDLIB=libunwind -DCMAKE_CROSSCOMPILING:BOOL=ON -DCROSS_TOOLCHAIN_FLAGS_NATIVE='-DLLDB_PYTHON_RELATIVE_PATH=/opt/yoe/2021.06-rc.1/sysroots/x86_64-yoesdk-linux/usr/lib/python3.9/site-packages -DCMAKE_TOOLCHAIN_FILE=/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0/toolchain-native.cmake' -DCMAKE_RANLIB=/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0/recipe-sysroot-native/usr/bin/x86_64-yoesdk-linux/x86_64-yoesdk-linux-llvm-ranlib -DCMAKE_AR=/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0/recipe-sysroot-native/usr/bin/x86_64-yoesdk-linux/x86_64-yoesdk-linux-llvm-ar -DCMAKE_NM=/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0/recipe-sysroot-native/usr/bin/x86_64-yoesdk-linux/x86_64-yoesdk-linux-llvm-nm -DCMAKE_STRIP=/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0/recipe-sysroot-native/usr/bin/x86_64-yoesdk-linux/x86_64-yoesdk-linux-llvm-strip -DLLVM_TABLEGEN=/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0/recipe-sysroot-native/usr/bin/llvm-tblgen -DCLANG_TABLEGEN=/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0/recipe-sysroot-native/usr/bin/clang-tblgen -DLLDB_TABLEGEN=/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0/recipe-sysroot-native/usr/bin/lldb-tblgen -DPYTHON_LIBRARY=/home/nice/develop/yoe/build/tmp/work/x86_64-nativesdk-yoesdk-linux/nativesdk-clang/13.0.0-r0/recipe-sysroot/opt/yoe/20 [... cut ...]
Checking whether the ASM compiler is IAR using "" did not match "IAR Assembler":
Checking whether the ASM compiler is ARMCC using "" did not match "(ARM Compiler)|(ARM Assembler)":
Checking whether the ASM compiler is NASM using "-v" did not match "(NASM version)":
Checking whether the ASM compiler is YASM using "--version" did not match "(yasm)":
Checking whether the ASM compiler is ADSP using "-version" did not match "Analog Devices":
Checking whether the ASM compiler is QCC using "-V" did not match "gcc_nto":

kraj, 2021-06-24:
yeah I think using clang-native to build nativesdk-clang is perhaps a good idea.

mr-nice, 2021-06-25:
Cool, I will try to provide a patch. Could take a bit

mr-nice, 2021-07-04:
I started pull request for a quick fix here https://github.com/kraj/meta-clang/pull/478 if someone has the same problem. Unfortunate I didn't found time to set the used compiler to clang-native

mr-nice, 2021-12-02:
I recompiled it and it works now, so we can close this issue?

kraj, 2021-12-02:
ok closing then

closed 2021-12-02 by kraj