YoeDistro/yoe-distro #572: odroid-c4 kernel build error

cbrake, 2021-09-02:
```
Currently  1 running tasks (853 of 3260)  26% |####################                                                           |                                
ERROR: linux-hardkernel-4.9.241+gitAUTOINC+20d79ab6a5-r0 do_kernel_metadata: Feature 'cfg/fs/vfat.scc' not found, this will cause configuration failures.      
ERROR: linux-hardkernel-4.9.241+gitAUTOINC+20d79ab6a5-r0 do_kernel_metadata: Check the SRC_URI for meta-data repositories or directories that may be missing   
ERROR: linux-hardkernel-4.9.241+gitAUTOINC+20d79ab6a5-r0 do_kernel_metadata: Set KERNEL_DANGLING_FEATURES_WARN_ONLY to ignore this issue                       
ERROR: linux-hardkernel-4.9.241+gitAUTOINC+20d79ab6a5-r0 do_kernel_metadata: Execution of '/scratch/yoe/yoe-distro/build/tmp/work/odroid_c4_hardkernel-yoe-linu
x/linux-hardkernel/4.9.241+gitAUTOINC+20d79ab6a5-r0/temp/run.do_kernel_metadata.1980791' failed with exit code 1:                                              
WARNING: /scratch/yoe/yoe-distro/build/tmp/work/odroid_c4_hardkernel-yoe-linux/linux-hardkernel/4.9.241+gitAUTOINC+20d79ab6a5-r0/temp/run.do_kernel_metadata.19
80791:404 exit 1 from 'exit 1'                                                                                                                                 
WARNING: Backtrace (BB generated script):                                                                                                                      
        #1: bbfatal_log, /scratch/yoe/yoe-distro/build/tmp/work/odroid_c4_hardkernel-yoe-linux/linux-hardkernel/4.9.241+gitAUTOINC+20d79ab6a5-r0/temp/run.do_ke
rnel_metadata.1980791, line 404                                                                                                                                
        #2: do_kernel_metadata, /scratch/yoe/yoe-distro/build/tmp/work/odroid_c4_hardkernel-yoe-linux/linux-hardkernel/4.9.241+gitAUTOINC+20d79ab6a5-r0/temp/ru
n.do_kernel_metadata.1980791, line 302                                                                                                                         
        #3: main, /scratch/yoe/yoe-distro/build/tmp/work/odroid_c4_hardkernel-yoe-linux/linux-hardkernel/4.9.241+gitAUTOINC+20d79ab6a5-r0/temp/run.do_kernel_me
tadata.1980791, line 435                                                                                                                                       
                                                                                                                                                               
Backtrace (metadata-relative locations):                                                                                                                       
        #1: bbfatal_log, /scratch/yoe/yoe-distro/sources/openembedded-core/meta/classes/logging.bbclass, line 72                                               
        #2: do_kernel_metadata, /scratch/yoe/yoe-distro/sources/openembedded-core/meta/classes/kernel-yocto.bbclass, line 258                                  
ERROR: Logfile of failure stored in: /scratch/yoe/yoe-distro/build/tmp/work/odroid_c4_hardkernel-yoe-linux/linux-hardkernel/4.9.241+gitAUTOINC+20d79ab6a5-r0/te
mp/log.do_kernel_metadata.1980791                                                                                                                              
Log data follows:                                   
```

kraj, 2021-09-02:
Can you try adding 

```
KERNEL_FEATURES:remove = "cfg/fs/vfat.scc"
```

via a bbappend or in main recipe ?

cbrake, 2021-09-02:
thanks @kraj -- that works.

Almost there, but getting the following now:

```

|   KSYM    .tmp_kallsyms2.o                                                   
|   LD      vmlinux                                                            
|   SORTEX  vmlinux                                                            
|   SYSMAP  System.map                                                         
|   OBJCOPY arch/arm64/boot/Image                                              
|   GZIP    arch/arm64/boot/Image.gz                                           
|   CHK     scripts/mod/devicetable-offsets.h                                  
| make[3]: *** No rule to make target 'arch/arm64/boot/dts/amlogic/meson-sm1-odroid-c4.dtb'.  Stop.
| make[2]: *** [arch/arm64/Makefile:148: amlogic/meson-sm1-odroid-c4.dtb] Error 2
| make[1]: *** [Makefile:152: sub-make] Error 2                                
| make: *** [Makefile:24: __sub-make] Error 2                                  
| WARNING: /scratch/yoe/yoe-distro/build/tmp/work/odroid_c4_hardkernel-yoe-linux/linux-hardkernel/4.9.241+gitAUTOINC+20d79ab6a5-r0/temp/run.do_compile.2059919:
307 exit 1 from 'exit 1'                                                       
| WARNING: Backtrace (BB generated script):                                    
|       #1: bbfatal_log, /scratch/yoe/yoe-distro/build/tmp/work/odroid_c4_hardkernel-yoe-linux/linux-hardkernel/4.9.241+gitAUTOINC+20d79ab6a5-r0/temp/run.do_co
mpile.2059919, line 307                                                        
|       #2: die, /scratch/yoe/yoe-distro/build/tmp/work/odroid_c4_hardkernel-yoe-linux/linux-hardkernel/4.9.241+gitAUTOINC+20d79ab6a5-r0/temp/run.do_compile.20
59919, line 228                                                                
|       #3: oe_runmake, /scratch/yoe/yoe-distro/build/tmp/work/odroid_c4_hardkernel-yoe-linux/linux-hardkernel/4.9.241+gitAUTOINC+20d79ab6a5-r0/temp/run.do_com
pile.2059919, line 214                                                         
|       #4: do_compile, /scratch/yoe/yoe-distro/build/tmp/work/odroid_c4_hardkernel-yoe-linux/linux-hardkernel/4.9.241+gitAUTOINC+20d79ab6a5-r0/temp/run.do_com
pile.2059919, line 156                                                         
|       #5: main, /scratch/yoe/yoe-distro/build/tmp/work/odroid_c4_hardkernel-yoe-linux/linux-hardkernel/4.9.241+gitAUTOINC+20d79ab6a5-r0/temp/run.do_compile.2
059919, line 311                                                               
|                                                                              
| Backtrace (metadata-relative locations):                                     
|       #1: bbfatal_log, /scratch/yoe/yoe-distro/sources/openembedded-core/meta/classes/logging.bbclass, line 72
|       #2: die, /scratch/yoe/yoe-distro/sources/openembedded-core/meta/classes/base.bbclass, line 56
|       #3: oe_runmake, /scratch/yoe/yoe-distro/sources/openembedded-core/meta/classes/base.bbclass, line 65
|       #4: do_compile, autogenerated, line 9                                  
ERROR: Task (/scratch/yoe/yoe-distro/sources/meta-odroid/recipes-kernel/linux/linux-hardkernel_4.9.bb:do_compile) failed with exit code '1'

```

kraj, 2021-09-03:
please try with updated pull which should fix the DT issue

closed 2021-09-03 by kraj