YoeDistro/yoe-distro #586: USB port is not working on BBB

cbrake, 2021-09-14:
First noticed that updates were not working from USB disk. Installed usbutils from feedserver and `lsusb` is returning nothing.

attached is kernel config.

[config.txt](https://github.com/YoeDistro/yoe-distro/files/7163211/config.txt)

kraj, 2021-09-14:
i see `CONFIG_USB=m` which means it will be part of the kernel modules perhaps turn that into `CONFIG_USB=y` and see if that helps

cbrake, 2021-09-14:
yes, after `opkg install kernel-modules`, USB now works.

5.10.59+git0+04b02a7e8e-0b.1 is the kernel version it is using.

from

sources/meta-ti/recipes-kernel/linux/linux-ti-staging_5.10.bb

ti33x/defconfig has the following contents:

`use-tisdk-config=ti_sdk_am3x_release`

The kernel source has a script titled: `ti_config_fragments/defconfig_builder`

Kernel source also has file named: `arch/arm/configs/ti_sdk_am3x_release_defconfig`

Seems to match what we have on the machine. 

Thoughts on the best way to fix this up?

kraj, 2021-11-22:
@cbrake are you seeing this with `yoe-initramfs-image` ? so when updater is triggered by kernel and updated are living on USB stick then it does not find it ?

cbrake, 2021-11-22:
@kraj yes, that is the main problem. We could add the kernel module to the main rootfs, but I've never used kernel modules much in the initramfs yet.

kraj, 2021-11-22:
I think we can define a kernel bbappend to add `CONFIG_USB=y` to kernel defconfig for all machines which support updater.

cbrake, 2021-11-22:
that sounds like a good solution

closed 2021-11-24 by cbrake