YoeDistro/yoe-distro #617: git:// github links in meta-raspberrypi

cbrake, 2021-11-17:
seeing the following when building raspberrypi3:

```
NOTE: Executing Tasks
WARNING: linux-firmware-rpidistro-20190114-2+rpt4-r0 do_fetch: Failed to fetch URL git://github.com/RPi-Distro/firmware-nonfree;branch=master;protocol=https, attempting MIRRORS if available
ERROR: linux-firmware-rpidistro-20190114-2+rpt4-r0 do_fetch: Fetcher failure: Unable to find revision 54ffdd6e2ea6055d46656b78e148fe7def3ec9d8 in branch master even from upstream
ERROR: linux-firmware-rpidistro-20190114-2+rpt4-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'git://github.com/RPi-Distro/firmware-nonfree;branch=master;protocol=https')
ERROR: Logfile of failure stored in: /scratch/yoe/yoe-distro/build/tmp/work/all-yoe-linux/linux-firmware-rpidistro/20190114-2+rpt4-r0/temp/log.do_fetch.24281
ERROR: Task (/scratch/yoe/yoe-distro/sources/meta-raspberrypi/recipes-kernel/linux-firmware-rpidistro/linux-firmware-rpidistro_git.bb:do_fetch) failed with exit code '1'
```

kraj, 2021-11-17:
yeah it seems upstream rebased the repo :(

kraj, 2021-11-17:
@cbrake can you try this patch out ?

closed 2021-11-18 by kraj

cbrake, 2021-11-18:
fetch now works -- thanks @kraj