YoeDistro/yoe-distro #638: Clang-native fails in do_compile on x84_64

mhill-holoplot, 2022-01-06:
I tried building postgresql with Yoe in order to debug this PR

However It seems the clang compilation fails on an eval. I haven't done anything strange or patched anything yet so I'm mildly confused as to why.

```
morgan@morgan-ubuntu:~/yoe$ bitbake postgresql
Setting MACHINE=qemux86-64
/home/morgan/yoe/conf/auto.conf has been updated
Setting DOCKER_REPO to yoedistro/yoe-build:buster
NOTE: Started PRServer with DBfile: /home/morgan/yoe/cache/prserv.sqlite3, Address: 127.0.0.1:45219, PID: 59
Loading cache: 100% |###############################################################################################################################################################################| Time: 0:00:01
Loaded 5148 entries from dependency cache.
Parsing recipes: 100% |#############################################################################################################################################################################| Time: 0:00:02
Parsing of 3565 .bb files complete (3564 cached, 1 parsed). 5151 targets, 683 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.53.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "x86_64-yoe-linux"
MACHINE              = "qemux86-64"
DISTRO               = "yoe"
DISTRO_VERSION       = "2021.12"
TUNE_FEATURES        = "m64 core2"
TARGET_FPU           = ""
meta-yoe             = "HEAD:258a2edbeb62525f7b03f6933ac5806c29c647f2"
meta-96boards        = "HEAD:c0a7f4052bd78f9a6972a5459840f639452bf76c"
meta-arm             
meta-arm-toolchain   = "HEAD:8061ecf0cde442cd5a36151c7dac91fcd0727e2d"
meta-atmel           = "HEAD:749ed98cd42c61234ad04476d5d8104f2087566c"
meta-freescale       = "HEAD:58a442bdafae4329a61f656d597c76c5fb8da928"
meta-freescale-3rdparty = "HEAD:b3670f9d5ffe1fd52f8c2d1cbb7027b9dcf79aad"
meta-odroid          
meta-odroid-extras   = "HEAD:e46880f21512622468a23c5c7fc60bf050cb63b0"
meta-qcom            = "HEAD:302e275d27dbe4bfbd2a8df8586b95d721750d63"
meta-raspberrypi     = "HEAD:58cc6625395c251e68604de587c332f07a6acd3b"
meta-riscv           = "HEAD:9561639c61663a10d8c9c23d26173db499f4c39b"
meta-rockchip        = "HEAD:17703ee37b46d15ec369588fbb86dde336df6028"
meta-ti              = "HEAD:75ff5168d39aa05cd918cfd36dd8695ee17a065d"
meta-clang           = "HEAD:e59b8f560a9e436f13588223c8f30c63d02ca3d8"
meta-elm-binary      = "HEAD:df225a0abb9f51c50714357d114d4d313b7bbe73"
meta-qt5             = "HEAD:b64b0a213240375e697185de06c3f502c65cdee7"
meta-filesystems     
meta-gnome           
meta-initramfs       
meta-multimedia      
meta-networking      
meta-oe              
meta-perl            
meta-python          
meta-webserver       
meta-xfce            = "HEAD:2b4f6a04f014da87a1f42bf67a52be59e5cd2b2e"
meta                 = "HEAD:24c1b8346a2ab8bdea2e140282e33814166d9113"

Initialising tasks: 100% |##########################################################################################################################################################################| Time: 0:00:01
Sstate summary: Wanted 103 Local 0 Mirrors 0 Missed 103 Current 210 (0% match, 67% complete)
NOTE: Executing Tasks
ERROR: clang-native-13.0.1-r0 do_compile: ExecutionError('/home/morgan/yoe/build/tmp/work/x86_64-linux/clang-native/13.0.1-r0/temp/run.do_compile.135', 1, None, None)
ERROR: Logfile of failure stored in: /home/morgan/yoe/build/tmp/work/x86_64-linux/clang-native/13.0.1-r0/temp/log.do_compile.135
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: cmake --build /home/morgan/yoe/build/tmp/work/x86_64-linux/clang-native/13.0.1-r0/build --target all --
| [1/5014] Building CXX object lib/CodeGen/CMakeFiles/LLVMCodeGen.dir/NonRelocatableStringpool.cpp.o
| [2/5014] Building CXX object lib/CodeGen/CMakeFiles/LLVMCodeGen.dir/PHIEliminationUtils.cpp.o
| [3/5014] Building CXX object lib/CodeGen/CMakeFiles/LLVMCodeGen.dir/OptimizePHIs.cpp.o
| [4/5014] Building CXX object lib/CodeGen/CMakeFiles/LLVMCodeGen.dir/ParallelCG.cpp.o
| [5/5014] Building CXX object lib/CodeGen/CMakeFiles/LLVMCodeGen.dir/PostRAHazardRecognizer.cpp.o
| [6/5014] Building CXX object lib/CodeGen/CMakeFiles/LLVMCodeGen.dir/PreISelIntrinsicLowering.cpp.o
| [7/5014] Building CXX object lib/CodeGen/CMakeFiles/LLVMCodeGen.dir/PostRASchedulerList.cpp.o
[... 123 lines cut ...]
| [118/5014] Building CXX object lib/CodeGen/AsmPrinter/CMakeFiles/LLVMAsmPrinter.dir/AsmPrinter.cpp.o
| [119/5014] Building CXX object lib/CodeGen/AsmPrinter/CMakeFiles/LLVMAsmPrinter.dir/DwarfExpression.cpp.o
| [120/5014] Building CXX object lib/CodeGen/AsmPrinter/CMakeFiles/LLVMAsmPrinter.dir/DwarfCompileUnit.cpp.o
| [121/5014] Building CXX object lib/CodeGen/AsmPrinter/CMakeFiles/LLVMAsmPrinter.dir/DwarfUnit.cpp.o
| [122/5014] Building CXX object lib/CodeGen/AsmPrinter/CMakeFiles/LLVMAsmPrinter.dir/DwarfDebug.cpp.o
| ninja: build stopped: subcommand failed.
| WARNING: /home/morgan/yoe/build/tmp/work/x86_64-linux/clang-native/13.0.1-r0/temp/run.do_compile.135:164 exit 1 from 'eval ${DESTDIR:+DESTDIR=${DESTDIR} } cmake --build '/home/morgan/yoe/build/tmp/work/x86_64-linux/clang-native/13.0.1-r0/build' "$@" -- ${EXTRA_OECMAKE_BUILD}'
| WARNING: Backtrace (BB generated script):
| 	#1: cmake_runcmake_build, /home/morgan/yoe/build/tmp/work/x86_64-linux/clang-native/13.0.1-r0/temp/run.do_compile.135, line 164
| 	#2: cmake_do_compile, /home/morgan/yoe/build/tmp/work/x86_64-linux/clang-native/13.0.1-r0/temp/run.do_compile.135, line 158
| 	#3: do_compile, /home/morgan/yoe/build/tmp/work/x86_64-linux/clang-native/13.0.1-r0/temp/run.do_compile.135, line 153
| 	#4: main, /home/morgan/yoe/build/tmp/work/x86_64-linux/clang-native/13.0.1-r0/temp/run.do_compile.135, line 177
ERROR: Task (virtual:native:/home/morgan/yoe/sources/meta-clang/recipes-devtools/clang/clang_git.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 914 tasks of which 903 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 5 seconds
NOTE: Build completion summary:
NOTE:   do_populate_sysroot: 0.0% sstate reuse(0 setscene, 1 scratch)
NOTE:   do_package: 0.0% sstate reuse(0 setscene, 1 scratch)
NOTE:   do_packagedata: 0.0% sstate reuse(0 setscene, 2 scratch)

Summary: 1 task failed:
  virtual:native:/home/morgan/yoe/sources/meta-clang/recipes-devtools/clang/clang_git.bb:do_compile
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

mhill-holoplot, 2022-01-06:
Umm I increased the amount of memory available to that VM from 8G to 16G and it seems to be have got further and not segfault-ed yet.

mhill-holoplot, 2022-01-06:
okay it built with more memory

closed 2022-01-06 by mhill-holoplot

cbrake, 2022-01-06:
thanks for reporting -- this is an important consideration with OE builds -- you need lots of memory!