YoeDistro/yoe-distro #639: Sanity check connection internet

caiortp, 2022-01-12:
scenario: Fresh Installed Ubuntu 20.04
Internet connection Ok
Docker installation OK
Docker + Internet OK

I was following the readme.md and trying to build an qemux86 yoe-simple-image.

```
$ bitbake yoe-simple-image
Setting MACHINE=qemux86-64
/home/cubo/workspace/yoe/conf/auto.conf has been updated
Setting DOCKER_REPO to yoedistro/yoe-build:buster
NOTE: Started PRServer with DBfile: /home/cubo/workspace/yoe/cache/prserv.sqlite3, Address: 127.0.0.1:45439, PID: 58
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    Fetcher failure for URL: 'https://www.example.com/'. URL https://www.example.com/ doesn't work.
    Please ensure your host's network is configured correctly.
    If your ISP or network is blocking the above URL,
    try with another domain name, for example by setting:
    CONNECTIVITY_CHECK_URIS = "https://www.yoctoproject.org/"    You could also set BB_NO_NETWORK = "1" to disable network
    access if all required sources are on local disk.

```
Solution:
Skip the connectivity_check in envsetup.sh adding the line below in the auto.conf creation.
CONNECTIVITY_CHECK_URIS = ""

kraj, 2022-05-02:
I think 0779158670ec531b669fb16ab98275fae53e0f70 fixed it.

kraj, 2022-05-02:
Please re-open if not.

closed 2022-05-02 by kraj