YoeDistro/yoe-distro #642: access tpm in yoe-initramfs-image

arrow53, 2022-01-18:
this isn't a bug, but just a question.

I'm including kernel module support for the slb9670 tpm module in my kernel. Booting into the `yoe-initramfs-image` recipe goes fine. Inside `updater` I'm going into a shell before `switch-root` for debugging.

If I include `kernel-modules` with this 

https://github.com/YoeDistro/yoe-distro/blob/master/sources/meta-yoe/recipes-core/images/yoe-initramfs-image.bb

I can then have access to `tpm_tis_spi.ko` inside the initramfs

So, once I boot into my initramfs I can then `modprobe tpm_tis_spi` and that works without errors.

The problem is there is no `/dev/tpm0` as it seems this is done later. If I go ahead and `switch_root` then later on I see the messages and the dev node created.
```
[   78.886814] tpm_tis_spi spi0.1: 2.0 TPM (device-id 0x1B, rev-id 22)
[   78.916629] tpm tpm0: A TPM error (256) occurred attempting the self test
[   78.916650] tpm tpm0: starting up the TPM manually
```

I'm pretty new to initramfs so maybe I'm doing something dumb. Is there something I can do to have the systemd load this in the initramfs image?

cbrake, 2022-01-18:
We typically don't run systemd (or any init system) in the initramfs to keep things light. However, you could change this, but I'd probably search for another solution first.

It seems when we use other devices, /dev/* entries are created automatically by the kernel. So, I'm not sure what is going on here. Perhaps there is another service that starts that fixes up the TPM device. If you can find this service, then you could extract the logic and add to your initramfs `init` file.

arrow53, 2022-01-18:
@cbrake that all makes sense. With Debian (stock raspberry pi) adding `tpm`, `tpm_tis_spi`, `tpm_tis_core` to `/etc/initramfs-tools/modules` does the trick and that of course has no systemd so seems what I'm doing should work. I can also just copy those files to a partition and manual load them with insmod or modprobe and I see the kernel messages and `/dev/tpm0` created.

Do device tree overlays not get loaded in the initramfs? I have an overlay for the tpm module so maybe that's it.

kraj, 2022-01-18:
@arrow53 our initramfs is primarily meant to boot into a mimimal system than can then either update the system or pivot into full booting mode. However if you just add tpm into kernel proper ( not as a module ) then it might work better with initramfs

arrow53, 2022-01-18:
thanks. Tried that first with no luck. Went with module loading to hopefully get more info. I'll go ahead and close as seems unrelated to yoe. Thanks!

closed 2022-01-18 by arrow53

cbrake, 2022-01-18:
> Do device tree overlays not get loaded in the initramfs? I have an overlay for the tpm module so maybe that's it.

Not that I know of, but it may be worth supporting that. I'm doing some work on overlays for another project (BBB), so would be interested in what you find.

Are you using the updater functionality in the initramfs?

arrow53, 2022-01-18:
> Are you using the updater functionality in the initramfs?

yes. i'm trying to integrate tpm and luks with it. bit of a rabbit hole. I think the tpm driver has some acpi / pnp stuff happening so the actual probe gets delayed until out of initramfs.

arrow53, 2022-01-19:
just to close this out in case anyone runs into it. If you enable the tpm in the kernel (not a module) and then also enable spi (not a module) the tpm does load. yeeash.

kraj, 2022-01-19:
Good hope you have fun using yoe !!