YoeDistro/yoe-distro #670: Tools missing

snathicktechno, 2022-04-14:
Hi,

Using below recipe, integrated openldap to riscv64qemu image
https://github.com/openembedded/meta-openembedded/blob/master/meta-oe/recipes-support/openldap/openldap_2.5.9.bb
Added below line in local.conf
IMAGE_INSTALL:append = " openldap"
qemuriscv64 image build successful.
missing SLAP tools (slapadd slapcat slapdn slapindex slappasswd slaptest slapauth slapacl slapschema slapmodify)

Please help me to resolve my issue.

Thanks

kraj, 2022-04-14:
you might want to try the same build on say qemuarm and see if you have these tools built, if not then perhaps some PACKAGECONFIGS are required to be enabled in the recipes. the default set maybe not enough to enable building these tools.

snathicktechno, 2022-04-15:
Hi KhemRaj,

As per your suggestion integrated openldap to qemuarm64 image and observed that tools are not built.

kraj, 2022-05-02:
@snathicktechno  thanks for confirming. I think we need to look into it further to root cause it.