YoeDistro/yoe-distro #675: change SIOT OS_VERSION_FIELD env var

cbrake, 2022-05-02:
For Yoe, it would be nice to set SIOT_OS_VERSION_FIELD to VERSION_ID, so it will pick up the Yoe version instead of the Yocto version.

https://github.com/simpleiot/simpleiot/blob/master/contrib/siot.service

kraj, 2022-05-03:
`os-release` is a standard artifact now a days, so perhaps we can read it and extract relevant information programmatically inside simpleIOT app itself perhaps

cbrake, 2022-05-03:
that is a good idea, so are you are suggesting we extract all the fields? -- is something like the below what you had in mind?

https://github.com/simpleiot/simpleiot/issues/358

cbrake, 2022-05-03:
information on `/etc/os-release`: 

https://www.freedesktop.org/software/systemd/man/os-release.html

closed 2022-05-05 by cbrake