YoeDistro/yoe-distro #711: SDK Image binutils do compile Error

collinbrake, 2022-09-07:
Running the following commands:
```
cd yoe
. ./rasberrypi3-64-envsetup.sh
yoe_setup
bitbake yoe-sdk-image > log.txt
```
[log.txt](https://github.com/YoeDistro/yoe-distro/files/9508431/log.txt)

Generates the attached log file.

collinbrake, 2022-09-07:
@cbrake I get a very similar error (related to `binutils do_compile()` when running:
```
bitbake yoe-simple-image -c populate_sdk
```

kraj, 2022-09-07:
@collinbrake I see this error burried in logs

```
| make[3]: *** [Makefile:1404: atomic.lo] Error 127
| make[3]: *** Waiting for unfinished jobs....

```

It seems your machine ran out of memory. Does it fix itself when you rerun the build ?

Also look into system logs on build machine to see if there were process crashes reported.

`coredumptctl` can show you core files.

kraj, 2022-09-07:
similar issue reported in gcc bugzilla

https://gcc.gnu.org/bugzilla/show_bug.cgi?format=multiple&id=100324

kraj, 2022-09-07:
interesting bit is

```
ls -l  /bin/sh
```

what does it show you.

collinbrake, 2022-09-07:
@kraj The above command shows me this:

```
........... yoe]$ ls -l /bin/sh lrwxrwxrwx 1 root root 4 Jan  8  2022 /bin/sh -> bash
```

kraj, 2022-09-07:
OK bash should work fine I guess. I also see  ` make -j 24` in logs. Perhaps try to limit parallelism. in `conf/local.conf` set

```
PARALLEL_MAKE = "-j 11"
BB_NUMBER_THREADS = "11"
XZ_THREADS = "8"
# Max 25% of memory to be used during parallel xz
XZ_MEMLIMIT = "10%"
```

cbrake, 2022-09-07:
that makes sense now -- I think this is a two cpu machine (so lots of cores) with 32GB of memory, so we do need to limit memory sometimes in these scenarios.

collinbrake, 2022-09-07:
[log2.txt](https://github.com/YoeDistro/yoe-distro/files/9510145/log2.txt)

Second run of:
```
bitbake yoe-simple-image -c populate_sdk
```

Resulted in a similar error, shown in log2.txt.

I will limit cores and give it a try. Thanks @kraj and @cbrake!

kraj, 2022-09-07:
> that makes sense now -- I think this is a two cpu machine (so lots of cores) with 32GB of memory, so we do need to limit threads sometimes in these scenarios.

I have observed that we need 2GB/core for not running into build eating all memory and invoking OOM killer, which then can start killing processes randomly.

collinbrake, 2022-09-07:
>OK bash should work fine I guess. I also see make -j 24 in logs. Perhaps try to limit parallelism. in conf/local.conf set

>PARALLEL_MAKE = "-j 11"
>BB_NUMBER_THREADS = "11"
>XZ_THREADS = "8"
>XZ_MEMLIMIT = "10%"

I created the `conf/local.conf` file with above content - it did not exist. Perhaps I need to direct Yocto to use this file since it did not exist before?

I am getting the same error as before: 
[log.do_populate_sdk.397](https://github.com/YoeDistro/yoe-distro/files/9515995/log.do_populate_sdk.397.txt)

I grepped the log file for `make -j`: `cat build/tmp/work/raspberrypi3_64-yoe-linux/yoe-simple-image/1.0-r0/temp/log.do_populate_sdk.397 | grep "make -j"`

and nothing was found.

Is it possible that there are cached results from the failure and re-running the build does not rerun the `make` command?

cbrake, 2022-09-07:
we added some notes on build machine requirements here: 

https://github.com/YoeDistro/yoe-distro/pull/712

cbrake, 2022-09-07:
> I created the `conf/local.conf` file with above content - it did not exist. Perhaps I need to direct Yocto to use this file since it did not exist before?

`local.conf` is an optional file for local user/machine settings (stuff you don't want checked into Git) and Yoe will use it if it exists. See https://github.com/YoeDistro/yoe-distro/blob/master/docs/conf-files.md

> Is it possible that there are cached results from the failure and re-running the build does not rerun the `make` command?

Anything is possible -- I sometimes move build to build-tmp and re-run the build.

I try to reproduce the problem here.

kraj, 2022-09-07:
> 

this is next error you are running into since you have made to `do_populate_sdk` means all needed packages are already built. 

You can `rm -rf build/tmp/`

and rebuild. It will use sstate and will be really quick.

kraj, 2022-09-07:
@collinbrake here is the interesting error in your logs

```
In * check_data_file_clashes: Package nativesdk-gprofng wants to install file /home/ventrac/yoe/build/tmp/work/raspberrypi3_64-yoe-linux/yoe-simple-image/1.0-r0/sdk/image/usr/local/oe-sdk-hardcoded-buildpath/sysroots/x86_64-yoesdk-linux/etc/gprofng.rc
        But that file is already provided by package  * binutils-cross-canadian-aarch64
```

which I think is a problem with gprogng support being added new in 2.39, 

Perhaps you can try this patch and report back.

https://github.com/YoeDistro/poky/commit/fdbcb35a56fc8cac888ba80ac1e5e1692185765f

```patch

From fdbcb35a56fc8cac888ba80ac1e5e1692185765f Mon Sep 17 00:00:00 2001
From: Khem Raj <raj.khem@gmail.com>
Date: Wed, 7 Sep 2022 16:10:40 -0700
Subject: [PATCH] binutils-cross-canadian: Do not package gprofng.rc

This file is already packaged via nativesdk-gprofng and can cause
populate SDK conflicts

---
 meta/recipes-devtools/binutils/binutils-cross-canadian.inc | 3 +--
 1 file changed, 1 insertion(+), 2 deletions(-)

diff --git a/meta/recipes-devtools/binutils/binutils-cross-canadian.inc b/meta/recipes-devtools/binutils/binutils-cross-canadian.inc
index 4e8f10c1c45..c18468ad50d 100644
--- a/meta/recipes-devtools/binutils/binutils-cross-canadian.inc
+++ b/meta/recipes-devtools/binutils/binutils-cross-canadian.inc
@@ -23,10 +23,9 @@ do_install () {
 	rm -f ${D}${libdir}/libiberty*
 	rm -f ${D}${libdir}/libopcodes*
 	rm -f ${D}${includedir}/*.h
+        rm -f ${D}${sysconfdir}/gprofng.rc
 	
 	cross_canadian_bindirlinks
 }
 
-FILES:${PN} += "${sysconfdir}/gprofng.rc"
-
 BBCLASSEXTEND = ""
```

collinbrake, 2022-09-08:
@kraj `build/tmp` must have been created by `bitbake yoe-simple-image -c populate_sdk` before it failed, because it exists. Ran mkdir command anyways with no change.

cbrake, 2022-09-08:
I'm getting the following here:

[log.do_populate_sdk.log](https://github.com/YoeDistro/yoe-distro/files/9529358/log.do_populate_sdk.log)

cbrake, 2022-09-08:
I'll try your patch now.

collinbrake, 2022-09-08:
Cloned a clean copy of `yoe-distro` and ran `cd sources/poky && git pull origin yoe/mut` to get the patch. Running bitbake now.

cbrake, 2022-09-08:
patch did not apply cleanly -- not sure why.

I'm getting:

```
Initialising tasks: 100% |#####################################################################################| Time: 0:00:02
Sstate summary: Wanted 9 Local 0 Mirrors 0 Missed 9 Current 1527 (0% match, 99% complete)
NOTE: Executing Tasks
ERROR: binutils-cross-canadian-aarch64-2.39-r0 do_package: QA Issue: binutils-cross-canadian-aarch64: Files/directories were installed but not shipped in any package:
  /usr/local/oe-sdk-hardcoded-buildpath/sysroots/x86_64-yoesdk-linux/etc
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
binutils-cross-canadian-aarch64: 1 installed and not shipped files. [installed-vs-shipped]
ERROR: binutils-cross-canadian-aarch64-2.39-r0 do_package: Fatal QA errors were found, failing task.
ERROR: Logfile of failure stored in: /scratch/yoe/yoe-distro/build/tmp/work/x86_64-nativesdk-yoesdk-linux/binutils-cross-canadian-aarch64/2.39-r0/temp/log.do_package.2616
ERROR: Task (/scratch/yoe/yoe-distro/sources/poky/meta/recipes-devtools/binutils/binutils-cross-canadian_2.39.bb:do_package) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4331 tasks of which 4324 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 6 seconds
NOTE: Build completion summary:
NOTE:   do_populate_sysroot: 0.0% sstate reuse(0 setscene, 1 scratch)
NOTE:   do_package: 0.0% sstate reuse(0 setscene, 1 scratch)

Summary: 1 task failed:
  /scratch/yoe/yoe-distro/sources/poky/meta/recipes-devtools/binutils/binutils-cross-canadian_2.39.bb:do_package
Summary: There were 2 ERROR messages, returning a non-zero exit code.
```

Ended up doing the following to get past it (probably not the final fix, but to get to the end):

```
diff --git a/meta/recipes-devtools/binutils/binutils-cross-canadian.inc b/meta/recipes-devtools/binutils/binutils-cross-canadian.inc
index 4e8f10c1c4..c6b4162dab 100644
--- a/meta/recipes-devtools/binutils/binutils-cross-canadian.inc
+++ b/meta/recipes-devtools/binutils/binutils-cross-canadian.inc
@@ -16,17 +16,18 @@ EXTRA_OECONF[vardepsexclude] = "TUNE_PKGARCH"
 do_install () {
        autotools_do_install
 
-       # We're not interested in the libs or headers, these would come from the 
+       # We're not interested in the libs or headers, these would come from the
        # nativesdk or target version of the binutils recipe
        rm -rf ${D}${prefix}/${TARGET_SYS}
        rm -f ${D}${libdir}/libbfd*
        rm -f ${D}${libdir}/libiberty*
        rm -f ${D}${libdir}/libopcodes*
        rm -f ${D}${includedir}/*.h
-       
+        rm -f ${D}${sysconfdir}/gprofng.rc
+        rm -rf ${D}${sysconfdir}
+
        cross_canadian_bindirlinks
 }
 
-FILES:${PN} += "${sysconfdir}/gprofng.rc"
```

kraj, 2022-09-08:
hmmm @cbrake you and @collinbrake are running into same problem. I think binutils patch should help your case. Yeah it may not not directly applicable on yoe since I did it on poky master but you can tweak it. 

I am not sure about the directory errors after deleting built/tmp that @collinbrake is getting. This is perhaps related to this directory being created by container or not. We need to look into this problem. I have hit it sometimes in past.

cbrake, 2022-09-08:
@collinbrake can you give me the results of the `id` command in your build account?

We've had some problems in the past if the UID is not 1000.

collinbrake, 2022-09-09:
> Cloned a clean copy of `yoe-distro` and ran `cd sources/poky && git pull origin yoe/mut` to get the patch. Running bitbake now.

The same error that @cbrake got resulted:

> patch did not apply cleanly -- not sure why.
> 
> After applying manually, I'm getting:
> 
> ```
> Initialising tasks: 100% |#####################################################################################| Time: 0:00:02
> Sstate summary: Wanted 9 Local 0 Mirrors 0 Missed 9 Current 1527 (0% match, 99% complete)
> NOTE: Executing Tasks
> ERROR: binutils-cross-canadian-aarch64-2.39-r0 do_package: QA Issue: binutils-cross-canadian-aarch64: Files/directories were installed but not shipped in any package:
>   /usr/local/oe-sdk-hardcoded-buildpath/sysroots/x86_64-yoesdk-linux/etc
> Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
> binutils-cross-canadian-aarch64: 1 installed and not shipped files. [installed-vs-shipped]
> ERROR: binutils-cross-canadian-aarch64-2.39-r0 do_package: Fatal QA errors were found, failing task.
> ERROR: Logfile of failure stored in: /scratch/yoe/yoe-distro/build/tmp/work/x86_64-nativesdk-yoesdk-linux/binutils-cross-canadian-aarch64/2.39-r0/temp/log.do_package.2616
> ERROR: Task (/scratch/yoe/yoe-distro/sources/poky/meta/recipes-devtools/binutils/binutils-cross-canadian_2.39.bb:do_package) failed with exit code '1'
> NOTE: Tasks Summary: Attempted 4331 tasks of which 4324 didn't need to be rerun and 1 failed.
> NOTE: Writing buildhistory
> NOTE: Writing buildhistory took: 6 seconds
> NOTE: Build completion summary:
> NOTE:   do_populate_sysroot: 0.0% sstate reuse(0 setscene, 1 scratch)
> NOTE:   do_package: 0.0% sstate reuse(0 setscene, 1 scratch)
> 
> Summary: 1 task failed:
>   /scratch/yoe/yoe-distro/sources/poky/meta/recipes-devtools/binutils/binutils-cross-canadian_2.39.bb:do_package
> Summary: There were 2 ERROR messages, returning a non-zero exit code.
> ```

Will try the fix that he used.

collinbrake, 2022-09-09:
> @collinbrake can you give me the results of the `id` command in your build account?
> 
> We've had some problems in the past if the UID is not 1000.

uid=1000(............

collinbrake, 2022-09-09:
@cbrake what file is that diff from? **???**/meta/recipes-devtools

I found recipes-devtools in meta-yoe but it has no `binutils` subfolder.

kraj, 2022-09-09:
> The same error that @cbrake got resulted:
>
>
> Will try the fix that he used.

the patch I posted was not quite right. Can you try the V2 of the patch from

https://github.com/YoeDistro/poky/commit/65e6f39851e49cb1b4d3f57db169880a67d93032

download the patch 

```
cd poky
curl -L -O https://github.com/YoeDistro/poky/commit/65e6f39851e49cb1b4d3f57db169880a67d93032.patch
git am 65e6f39851e49cb1b4d3f57db169880a67d93032.patch
```

then do the build again.

cbrake, 2022-09-09:
@kraj your new patch works here on latest Yoe master.

Thanks!

kraj, 2022-09-09:
I have posted a pull request for yoe-distro https://github.com/YoeDistro/yoe-distro/pull/713 once its merged. The binutils issu should be addressed.

cbrake, 2022-09-12:
@kraj this patch does not appear to be in the latest yoe master -- am I missing something?

kraj, 2022-09-12:
What’s the commit hash of poky repo

cbrake, 2022-09-12:
df89d59a19f3000d829ee6b18364260a4e5312e8

cbrake, 2022-09-12:
looks like your change is in upstream poky:

https://git.yoctoproject.org/poky/log/

kraj, 2022-09-12:
Yeah but it’s one commit after the current poky pinned in yoe master :) 

I will soon create a pull request to update layers perhaps later today which should also bring this in

cbrake, 2022-09-13:
`bitbake yoe-simple-image -c populate_sdk` works in latest yoe/master -- thanks @kraj 

I'll try yoe-sdk-image later today just for good measure.

cbrake, 2022-09-13:
this builds now, closing. Thanks @kraj!

closed 2022-09-13 by cbrake