YoeDistro/yoe-distro #729: get C4 build working again

cbrake, 2022-10-13:
need to work through a few kernel build issues:

```
| /scratch/yoe/yoe-distro/build/tmp/work-shared/odroid-c4-hardkernel/kernel-source/drivers/amlogic/media/vid
eo_processor/ionvideo/ppmgr2.c: In function 'ppmgr2_canvas_config':                                         
| /scratch/yoe/yoe-distro/build/tmp/work-shared/odroid-c4-hardkernel/kernel-source/drivers/amlogic/media/vid
eo_processor/ionvideo/ppmgr2.c:392:13: error: the comparison will always evaluate as 'true' for the address 
of 'phy_addr' will never be NULL [-Werror=address]    
|   392 |         if (!ppd->phy_addr) {               
|       |             ^                                                                                     
| In file included from /scratch/yoe/yoe-distro/build/tmp/work-shared/odroid-c4-hardkernel/kernel-source/dri
vers/amlogic/media/video_processor/ionvideo/ppmgr2.c:18:                                                    
| /scratch/yoe/yoe-distro/build/tmp/work-shared/odroid-c4-hardkernel/kernel-source/drivers/amlogic/media/vid
eo_processor/ionvideo/ionvideo.h:211:15: note: 'phy_addr' declared here                                     
|   211 |         void *phy_addr[PPMGR2_MAX_CANVAS];                                                        
|       |               ^~~~~~~~                                                                            | cc1: all warnings being treated as errors                                                                 
| make[7]: *** [/scratch/yoe/yoe-distro/build/tmp/work-shared/odroid-c4-hardkernel/kernel-source/scripts/Mak
efile.build:335: drivers/amlogic/media/video_processor/ionvideo/ppmgr2.o] Error 1
| make[6]: *** [/scratch/yoe/yoe-distro/build/tmp/work-shared/odroid-c4-hardkernel/kernel-source/scripts/Mak
efile.build:648: drivers/amlogic/media/video_processor/ionvideo] Error 2                                    
| make[6]: *** Waiting for unfinished jobs....                                     

```

kraj, 2022-10-13:
what kernel version is it trying to build ?

cbrake, 2022-10-13:
ERROR: Task (/scratch/yoe/yoe-distro/sources/meta-odroid/recipes-kernel/linux/linux-hardkernel_4.9.bb:do_compile) failed with exit code '1'

kraj, 2022-10-13:
yeah I will cook a fix, I think its something that gcc12 is finding in old codebase, we can perhaps disable the concerned warning.

kraj, 2022-10-13:
Please try the fixes posted in https://github.com/akuster/meta-odroid/pull/86

kraj, 2023-04-11:
I think we have this working now in Yoe, if it does not then please re-open.

closed 2023-04-11 by kraj