YoeDistro/yoe-distro #786: variscite kernel patch problem

cbrake, 2023-04-28:
```
 DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_patch
| (1/2) 0001-init_disassemble_info-signature-changes-causes-compi.patch
| [INFO]: check of .kernel-meta//patches//./0001-init_disassemble_info-signature-changes-causes-compi.patch with "git am" did not pass, trying reduced context.
| (2/2) 0001-perf-tools-Add-an-option-to-build-without-libbfd.patch
| [INFO]: check of .kernel-meta//patches//./0001-perf-tools-Add-an-option-to-build-without-libbfd.patch with "git am" did not pass, trying reduced context.
| [INFO]: Context reduced git-am of .kernel-meta//patches//./0001-perf-tools-Add-an-option-to-build-without-libbfd.patch with "git am" did not work, trying "apply".
| error: patch failed: tools/perf/Makefile.config:805
| error: tools/perf/Makefile.config: patch does not apply
| [ERROR]: Application of .kernel-meta//patches//./0001-perf-tools-Add-an-option-to-build-without-libbfd.patch failed.
|          Patch needs to be refreshed. Sample resolution script:
|              .git/rebase-apply/resolve_rejects
| ERROR: Could not apply patches for imx8qm-var-som.
| ERROR: Patch failures can be resolved in the linux source directory /scratch/yoe/yoe-distro/build/tmp/work/imx8qm_var_som-yoe-linux/linux-variscite/5.15.60+gitAUTOINC+84a7bb1736-r0/git)
| WARNING: /scratch/yoe/yoe-distro/build/tmp/work/imx8qm_var_som-yoe-linux/linux-variscite/5.15.60+gitAUTOINC+84a7bb1736-r0/temp/run.do_patch.100940:207 exit 1 from 'exit 1'
| WARNING: Backtrace (BB generated script):
|       #1: bbfatal_log, /scratch/yoe/yoe-distro/build/tmp/work/imx8qm_var_som-yoe-linux/linux-variscite/5.15.60+gitAUTOINC+84a7bb1736-r0/temp/run.do_patch.100940, line 207
|       #2: do_patch, /scratch/yoe/yoe-distro/build/tmp/work/imx8qm_var_som-yoe-linux/linux-variscite/5.15.60+gitAUTOINC+84a7bb1736-r0/temp/run.do_patch.100940, line 161
|       #3: main, /scratch/yoe/yoe-distro/build/tmp/work/imx8qm_var_som-yoe-linux/linux-variscite/5.15.60+gitAUTOINC+84a7bb1736-r0/temp/run.do_patch.100940, line 230
ERROR: Task (/scratch/yoe/yoe-distro/sources/meta-variscite-bsp/recipes-kernel/linux/linux-variscite_5.10.72.bb:do_patch) failed with exit code '1'
```

If I restart the build, it seems to keep going.

cbrake, 2023-04-28:
it is building `linux-variscite-5.15.60+gitAUTOINC+84a7bb1736-r0`, but the recipes in the sources are:

```
[cbrake@ceres sources]$ find -name linux-variscite*
./meta-freescale-3rdparty/recipes-kernel/linux/linux-variscite
./meta-freescale-3rdparty/recipes-kernel/linux/linux-variscite_4.14.bb
./meta-yoe/dynamic-layers/variscite-bsp/recipes-kernel/linux/linux-variscite
./meta-yoe/dynamic-layers/variscite-bsp/recipes-kernel/linux/linux-variscite_%.bbappend
./meta-variscite-bsp/recipes-kernel/linux/linux-variscite_5.10.72.bb
./meta-variscite-bsp/recipes-kernel/linux/linux-variscite
```

kraj, 2023-04-29:
https://github.com/YoeDistro/yoe-distro/pull/785 Should fix it

cbrake, 2023-04-29:
showing the 5.15 recipe now:

```
[cbrake@ceres sources]$ find -name linux-variscite*
./meta-freescale-3rdparty/recipes-kernel/linux/linux-variscite
./meta-freescale-3rdparty/recipes-kernel/linux/linux-variscite_4.14.bb
./meta-yoe/dynamic-layers/variscite-bsp/recipes-kernel/linux/linux-variscite
./meta-yoe/dynamic-layers/variscite-bsp/recipes-kernel/linux/linux-variscite_%.bbappend
./meta-variscite-bsp/recipes-kernel/linux/linux-variscite_5.15.bb
```

cbrake, 2023-04-29:
and do_patch seems to be working now.

cbrake, 2023-04-29:
`imx8qm-var-som` builds now -- trying `imx8qxp-var-som` again ...

cbrake, 2023-04-29:
`imx8qxp-var-som` builds - now -- testing installer/updater ...

cbrake, 2023-04-29:
Seeing the following in u-boot:

```
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc1 is current device
Failed to load '/boot/boot.scr'
Failed to load '/boot/Image.gz'
Error: Bad gzipped data
Booting from net ...
ethernet@5b040000 Waiting for PHY auto negotiation to complete......................................... TIMEOUT !
```

it does not appear that bootdir is getting modified in the u-boot env.

Patched the bootloader to fix the bootdir path, now seeing:

Failed to load '/imx8qxp-var-som-symphony.dtb'

Apparently we need to copy the correct dtb to this file.

cbrake, 2023-04-29:
I tried building the imx8qm-var-som machine -- figured it would at least boot on the board I have -- seeing the same issue:

`Failed to load '/imx8qxp-var-som-symphony.dtb'`

@kraj do you know how they select the correct DTB on boot? It seems they need to detect on boot if they are booting from SD or eMMC and then select the correct DTB, but maybe that is set during build time.

cbrake, 2023-05-02:
Took a look at the Variscite rescue SD card, and it soft links imx8qxp-var-som-symphony.dtb, so need to figure out how to do that:

[image]

kraj, 2023-05-02:
ah see how we do the aliasing for default DTB for `imx8qm-var-som` you need something similar.

https://github.com/YoeDistro/yoe-distro/blob/master/sources/meta-yoe/recipes-core/images/machines/imx8qm-var-som.inc#L30

and

https://github.com/YoeDistro/yoe-distro/blob/master/sources/meta-yoe/recipes-core/images/machines/imx8qm-var-som.inc#L47

hope that helps.

cbrake, 2023-05-02:
yeah, I tried to copy that:

https://github.com/YoeDistro/yoe-distro/pull/787/files#diff-9cb956f234d2dbe5391adcccc4d5f9e0c417ec76841ba30f723d8a9477c56231

For some reason, the `imx8qxp-var-som-symphony.dtb` file is not ending up on the boot partition. 

I also tried booting the same `imx8qm-var-som` machine with the same result, so curious if that machine currently boots for you?

kraj, 2023-05-02:
is it in project config.conf see

https://github.com/YoeDistro/yoe-distro/blob/master/sources/meta-yoe/conf/projects/var-som-mx8/config.conf#L23

we use lvds one for the symphony board I have.

cbrake, 2023-05-03:
OK, installer is booting now, but need to get the updater to write the dtb to the eMMC boot partition.

The problem I was facing when testing the imx8qm-var-som was that I was still booting from eMMC :-(

Almost there ...

cbrake, 2023-05-03:
it boots:

`Yoe Linux 2023.04 imx8qxp-var-som ttyLP3`

Getting the following on boot:

`[FAILED] Failed to start Variscite WIFI setup service.`

If I look at boot, the dtb is soft linking to the sd one instead of wifi:

`imx8qxp-var-som-symphony.dtb -> imx8qxp-var-som-symphony-sd.dtb`

Tomorrows problem ...

kraj, 2023-05-03:
To use wifi DTB as default I guess in `config.conf` you need 

`IMAGE_BOOT_FILES:append = " imx8qxp-var-som-symphony-wifi.dtb;imx8qxp-var-som-symphony.dtb"`

cbrake, 2023-05-03:
some how go eMMC in a bad state:

```
Updater version 20.5
VAR-SOM-MX8 Symphony Board v20.5
10
Running imx8qxp-var-som initializations ...
updater version 20.5
Sleeping for 4 second(s) for USB flash to settle...
checking usb for update ...
No USB Installer disk detected
checking sd for update ...
[    8.589902] mmc1: tried to HW reset card, got error -110
[    8.595273] mmcblk1: recovery failed!
[    8.598989] blk_update_request: I/O error, dev mmcblk1, sector 8194 op 0x0:(READ) flags 0x1000 phys_seg 1 prio class 0
[    8.609760] EXT4-fs (mmcblk1p1): unable to read superblock
[    8.620719] mmcblk1: recovery failed!
[    8.624416] blk_update_request: I/O error, dev mmcblk1, sector 8194 op 0x0:(READ) flags 0x1000 phys_seg 1 prio class 0
[    8.635152] EXT4-fs (mmcblk1p1): unable to read superblock
[    8.645967] mmcblk1: recovery failed!
[    8.649654] blk_update_request: I/O error, dev mmcblk1, sector 8194 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[    8.660131] EXT2-fs (mmcblk1p1): error: unable to read superblock
[    8.671560] mmcblk1: recovery failed!
[    8.675260] blk_update_request: I/O error, dev mmcblk1, sector 8192 op 0x0:(READ) flags 0x800 phys_seg 1 prio class 0
[    8.685904] SQUASHFS error: Failed to read block 0x0: -5
[    8.691256] unable to read squashfs_super_block
[    8.701049] mmcblk1: recovery failed!
[    8.704754] blk_update_request: I/O error, dev mmcblk1, sector 8192 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[    8.715225] FAT-fs (mmcblk1p1): unable to read boot sector
No SD card detected
checking data partition for update ...
Mounting data partition
Checking data partition size ...
cat: can't open '/sys/class/block/mmcblk0p3/size': No such file or directory
sh: 2048: unknown operand
Resizing data partition ...
warning: /dev/mmcblk0: partition 3 is not defined yet
Checking that no-one is using this disk right now ... OK

Disk /dev/mmcblk0: 14.56 GiB, 15634268160 bytes, 30535680 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xb64143a2

Old situation:

Device         Boot Start      End  Sectors  Size Id Type
/dev/mmcblk0p1       2048 30535679 30533632 14.6G 83 Linux

/dev/mmcblk0p3: No free sectors available.
Leaving.

resize2fs 1.47.0 (5-Feb-2023)
open: No such file or directory while opening /dev/mmcblk0p3
Resizing emmc failed, bad media

Something went wrong. Dropping you to a shell.
--install: applet not found
/bin/sh: can't access tty; job control turned off
~ # 

~ # ls /dev/mmcblk
mmcblk0       mmcblk0boot1  mmcblk0rpmb   mmcblk1p1
mmcblk0boot0  mmcblk0p1     mmcblk1
```

I then delete all the partitions on eMMC with fdisk and got:

```
No SD card detected
checking data partition for update ...
Mounting data partition
Checking data partition size ...
cat: can't open '/sys/class/block/mmcblk0p3/size': No such file or directory
sh: 2048: unknown operand
Resizing data partition ...
warning: /dev/mmcblk0: partition 3 is not defined yet
Checking that no-one is using this disk right now ... OK

Disk /dev/mmcblk0: 14.56 GiB, 15634268160 bytes, 30535680 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xb64143a2

Old situation:

/dev/mmcblk0p3: Created a new partition 1 of type 'Linux' and of size 14.6 GiB.

New situation:
Disklabel type: dos
Disk identifier: 0xb64143a2

Device         Boot Start      End  Sectors  Size Id Type
/dev/mmcblk0p1       2048 30535679 30533632 14.6G 83 Linux

The partition table has been altered.
Calling ioctl() to re-read partition table.[    9.040833]  mmcblk0: p1

Syncing disks.
resize2fs 1.47.0 (5-Feb-2023)
open: No such file or directory while opening /dev/mmcblk0p3
Resizing emmc failed, bad media

Something went wrong. Dropping you to a shell.
--install: applet not found
/bin/sh: can't access tty; job control turned off
~ # 
```

cbrake, 2023-05-03:
so I reverted the following change, and installer runs again:

[image]

As a test, I manually re-linked the dtb to the wifi version, and wifi still fails on boot.

After several minutes, it still does not show up.

Still digging ...

closed 2023-08-02 by cbrake