YoeDistro/yoe-distro #788: permission denied build/tmp/hosttools

cbrake, 2023-04-29:
I'm running into an issues where if I `rm -rf build/tmp` and then restart a build, I get:

```
[cbrake@ceres yoe-clean-test]$ bitbake yoe-simple-image
usermod: no changes
Setting PROJECT=rpi4-64
Setting MACHINE=raspberrypi4-64
/scratch/yoe/yoe-clean-test/conf/auto.conf has been updated
Setting DOCKER_REPO to yoedistro/yoe-build:bullseye-x86_64
Traceback (most recent call last):
  File "/scratch/yoe/yoe-clean-test/sources/poky/bitbake/bin/bitbake", line 36, in <module>
    sys.exit(bitbake_main(BitBakeConfigParameters(sys.argv),
  File "/scratch/yoe/yoe-clean-test/sources/poky/bitbake/lib/bb/main.py", line 404, in bitbake_main
    return ui_module.main(server_connection.connection, server_connection.events,
  File "/scratch/yoe/yoe-clean-test/sources/poky/bitbake/lib/bb/ui/knotty.py", line 415, in main
    params.updateToServer(server, os.environ.copy())
  File "/scratch/yoe/yoe-clean-test/sources/poky/bitbake/lib/bb/cookerdata.py", line 75, in updateToServer
    raise Exception("Unable to update the server configuration with local parameters: %s" % error)
Exception: Unable to update the server configuration with local parameters: Traceback (most recent call last):
  File "/scratch/yoe/yoe-clean-test/sources/poky/bitbake/lib/bb/command.py", line 91, in runCommand
    result = command_method(self, commandline)
  File "/scratch/yoe/yoe-clean-test/sources/poky/bitbake/lib/bb/command.py", line 290, in updateConfig
    command.cooker.updateConfigOpts(options, environment, cmdline)
  File "/scratch/yoe/yoe-clean-test/sources/poky/bitbake/lib/bb/cooker.py", line 552, in updateConfigOpts
    self.reset()
  File "/scratch/yoe/yoe-clean-test/sources/poky/bitbake/lib/bb/cooker.py", line 1786, in reset
    self.initConfigurationData()
  File "/scratch/yoe/yoe-clean-test/sources/poky/bitbake/lib/bb/cooker.py", line 409, in initConfigurationData
    self.databuilder.parseBaseConfiguration()
  File "/scratch/yoe/yoe-clean-test/sources/poky/bitbake/lib/bb/cookerdata.py", line 267, in parseBaseConfiguration
    bb.event.fire(bb.event.ConfigParsed(), self.data)
  File "/scratch/yoe/yoe-clean-test/sources/poky/bitbake/lib/bb/event.py", line 225, in fire
    fire_class_handlers(event, d)
  File "/scratch/yoe/yoe-clean-test/sources/poky/bitbake/lib/bb/event.py", line 132, in fire_class_handlers
    execute_handler(name, handler, event, d)
  File "/scratch/yoe/yoe-clean-test/sources/poky/bitbake/lib/bb/event.py", line 103, in execute_handler
    ret = handler(event, d)
  File "/scratch/yoe/yoe-clean-test/sources/poky/meta/classes-global/base.bbclass", line 261, in defaultbase_eventhandler
    setup_hosttools_dir(d.getVar('HOSTTOOLS_DIR'), 'HOSTTOOLS', d)
  File "/scratch/yoe/yoe-clean-test/sources/poky/meta/classes-global/base.bbclass", line 106, in setup_hosttools_dir
    bb.utils.mkdirhier(dest)
  File "/scratch/yoe/yoe-clean-test/sources/poky/bitbake/lib/bb/utils.py", line 767, in mkdirhier
    raise e
  File "/scratch/yoe/yoe-clean-test/sources/poky/bitbake/lib/bb/utils.py", line 764, in mkdirhier
    os.makedirs(directory)
  File "/usr/lib/python3.9/os.py", line 225, in makedirs
    mkdir(name, mode)
PermissionError: [Errno 13] Permission denied: '/scratch/yoe/yoe-clean-test/build/tmp/hosttools'
```

This is using a pre-existing sstate-cache.

It appears build/tmp is getting created with root permissions:

```
lrwxrwxrwx 1 cbrake cbrake   18 Apr 29 15:02 sstate-cache -> ../../sstate-cache
drwxr-xr-x 2 root   root   4096 Apr 29 15:03 tmp
```

If I chown back to me, then build works fine.

Also, I only see this when building in docker -- unsetting DOCKER_REPO fixes this problem.

kraj, 2023-04-30:
my `tmp` is mapped as bind mount so I did not see this issue locally, let me try to reproduce this here

kraj, 2023-04-30:
On arm builder I tried 
```
rm -rf build/tmp
```
exit the shell and log back in.

```
.khem@openembedded:~/yoe$ . ./envsetup.sh qemuarm64
Setting PROJECT=qemuarm64
Setting MACHINE=qemuarm64
/home/khem/yoe/conf/auto.conf has been updated

$ bitbake yoe-simple-image
```

and it works fine.

kraj, 2023-04-30:
Can you delete your container image and let it download latest ?