YoeDistro/yoe-distro #792: xwayland do_patch failure

cbrake, 2023-05-05:
rebased cbrake/master (variscite changes) on latest and attempting to build yoe-kiosk-image (to get wayland)

```
ERROR: xwayland-23.1.1-r0 do_patch: Applying patch '0002-glamor-Fix-fbo-pixmap-format-with-GL_BGRA_EXT.patch' on target directory '/scratch/yoe/yoe-distro/build/tmp/work/cortexa35-mx8-yoe-linux/xwayland/23.1.1-r0/xwayland-23.1.1'
CmdError('quilt --quiltrc /scratch/yoe/yoe-distro/build/tmp/work/cortexa35-mx8-yoe-linux/xwayland/23.1.1-r0/recipe-sysroot-native/etc/quiltrc push', 0, 'stdout: Applying patch 0002-glamor-Fix-fbo-pixmap-format-with-GL_BGRA_EXT.patch
patching file glamor/glamor.c
Hunk #1 FAILED at 586.
1 out of 1 hunk FAILED -- rejects in file glamor/glamor.c
Patch 0002-glamor-Fix-fbo-pixmap-format-with-GL_BGRA_EXT.patch does not apply (enforce with -f)

stderr: ')
ERROR: Logfile of failure stored in: /scratch/yoe/yoe-distro/build/tmp/work/cortexa35-mx8-yoe-linux/xwayland/23.1.1-r0/temp/log.do_patch.2424
ERROR: Task (/scratch/yoe/yoe-distro/sources/poky/meta/recipes-graphics/xwayland/xwayland_23.1.1.bb:do_patch) failed with exit code '1'
```

cbrake, 2023-05-05:
looks like a meta-freescale patch:

`./meta-freescale/recipes-graphics/xwayland/xwayland/0002-glamor-Fix-fbo-pixmap-format-with-GL_BGRA_EXT.patch`

kraj, 2023-05-05:
yeah we need this https://github.com/Freescale/meta-freescale/pull/1489
which is now merged in upstream meta-freescale. So we need to upgrade master yoe-distro to bump this layer to latest

kraj, 2023-05-05:
you can upgrade to latest `master` of yoe-distro and it should have the needed fix.

cbrake, 2023-05-05:
thanks @kraj -- testing now!

closed 2023-05-05 by cbrake