YoeDistro/yoe-distro #817: qtwebengine build failure

cbrake, 2023-08-02:
Getting the following with the latest yoe master:

```
| In file included from ../../../../../git/src/3rdparty/chromium/third_party/skia/src/core/SkOpts.cpp:46:
| ../../../../../git/src/3rdparty/chromium/third_party/skia/src/opts/SkRasterPipeline_opts.h:1005:5: warning: 'memcpy' will always overflow; destination buffer has size 2, but size argument is 8 [-Wfortify-source]
|     memcpy(&fp16, &h, sizeof(U16));
|     ^
| ../../../../../git/src/3rdparty/chromium/third_party/skia/src/opts/SkRasterPipeline_opts.h:1030:19: error: functional-style cast from 'F' (aka 'V<float>') to '__fp16' is not allowed
|     __fp16 fp16 = __fp16(f);
|                   ^~~~~~~~
| 1 warning and 1 error generated.
| [1706/30030] CXX obj/skia/skia_core_and_effects/SkPaintPriv.o
| [1707/30030] CXX obj/skia/skia_core_and_effects/SkQuadClipper.o
| [1708/30030] CXX obj/skia/skia_core_and_effects/SkModeColorFilter.o
| [1709/30030] CXX obj/skia/skia_core_and_effects/SkPipelineData.o
| [1710/30030] CXX obj/skia/skia_core_and_effects/SkPaintParamsKey.o
| [1711/30030] CXX obj/skia/skia_core_and_effects/SkRTree.o
| [1712/30030] CXX obj/skia/skia_core_and_effects/SkPromiseImageTexture.o
| [1713/30030] CXX obj/skia/skia_core_and_effects/SkPaint.o
| [1714/30030] CXX obj/skia/skia_core_and_effects/SkOverdrawCanvas.o
| [1715/30030] CXX obj/skia/skia_core_and_effects/SkRRect.o
| [1716/30030] CXX obj/skia/skia_core_and_effects/SkPathEffect.o
| [1717/30030] CXX obj/skia/skia_core_and_effects/SkRasterClip.o
| [1718/30030] CXX obj/skia/skia_core_and_effects/SkPixmap.o
| [1719/30030] CXX obj/skia/skia_core_and_effects/SkRecord.o
| [1720/30030] CXX obj/skia/skia_core_and_effects/SkRegion.o
| [1721/30030] CXX obj/skia/skia_core_and_effects/SkRasterPipeline.o
| [1722/30030] CXX obj/skia/skia_core_and_effects/SkRecords.o
| [1723/30030] CXX obj/skia/skia_core_and_effects/SkPath.o
| [1724/30030] CXX obj/skia/skia_core_and_effects/SkRect.o
| [1725/30030] CXX obj/skia/skia_core_and_effects/SkMipmap.o
| [1726/30030] CXX obj/skia/skia_core_and_effects/SkMesh.o
| [1727/30030] CXX obj/skia/skia_core_and_effects/SkRecordOpts.o
| [1728/30030] CXX obj/skia/skia_core_and_effects/SkRasterPipelineBlitter.o
| [1729/30030] CXX obj/skia/skia_core_and_effects/SkRecordDraw.o
| [1730/30030] CXX obj/skia/skia_core_and_effects/SkReadBuffer.o
| ninja: build stopped: subcommand failed.
| [98/290] Linking CXX shared module lib/qml/QtWebEngine/ControlsDelegates/libqtwebenginequickdelegatesplugin.so
| FAILED: src/core/Release/aarch64/QtWebEngineCore.stamp src/core/Release/aarch64/QtWebEngineCore /scratch/yoe/yoe-distro/build/tmp/work/cortexa53-crypto-mx8mn-yoe-linux/qtwebengine/6.5.3-r0/build/src/core/Release/aarch64/QtWebEngineCore.stamp /scratch/yoe/yoe-distro/build/tmp/work/cortexa53-crypto-mx8mn-yoe-linux/qtwebengine/6.5.3-r0/build/src/core/Release/aarch64/QtWebEngineCore
| cd /scratch/yoe/yoe-distro/build/tmp/work/cortexa53-crypto-mx8mn-yoe-linux/qtwebengine/6.5.3-r0/build/src/core && /scratch/yoe/yoe-distro/build/tmp/work/cortexa53-crypto-mx8mn-yoe-linux/qtwebengine/6.5.3-r0/recipe-sysroot-native/usr/bin/ninja -C /scratch/yoe/yoe-distro/build/tmp/work/cortexa53-crypto-mx8mn-yoe-linux/qtwebengine/6.5.3-r0/build/src/core/Release/aarch64 QtWebEngineCore
| ninja: build stopped: subcommand failed.
| WARNING: exit code 1 from a shell command.
ERROR: Task (/scratch/yoe/yoe-distro/sources/meta-qt6/recipes-qt/qt6/qtwebengine_git.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 5349 tasks of which 2760 didn't need to be rerun and 1 failed.
```

kraj, 2023-08-02:
We need to use `yoe/mut` layer for `meta-qt6` for now. Our fixes are yet not merged upstream.

cbrake, 2023-08-02:
thanks, that works. Do you think we should pin the meta-qt6 submodule to yoe/mut for now, or will that cause other problems?

kraj, 2023-08-02:
> thanks, that works. Do you think we should pin the meta-qt6 submodule to yoe/mut for now, or will that cause other problems?

`yoe/mut` is an integration/staging branch so it gets rebased as soon as upstream have updates to `6.5` branch
I think we should create a new branch `yoe/6.5` from yoe/mut and then do merges to it instead of rebase

cbrake, 2023-08-02:
does it matter if we rebase a submodule, as submodules pin to a hash, so submodule updates should work.

I guess the problem occurs if the hash disappears from the upstream, so not sure how that works.

kraj, 2023-08-02:
> does it matter if we rebase a submodule, as submodules pin to a hash, so submodule updates should work.
> 

It might probably after garbage collection is run and github might be doing it in background.

> I guess the problem occurs if the hash disappears from the upstream, so not sure how that works.

kraj, 2023-09-29:
This has been fixed in meta-qt6 now.

closed 2023-09-29 by kraj