YoeDistro/yoe-distro #829: updater rescue_shell busybox install

cbrake, 2023-08-23:
we have the following code in the update:

``` sh
rescue_shell() {
  msg "$@"
  msg "Something went wrong. Dropping you to a shell."
  busybox --install -s
  exec /bin/sh
}
```

When the shell is launched, we get:

```
Something went wrong. Dropping you to a shell.
--install: applet not found
/bin/sh: can't access tty; job control turned off
~ # 
~ #                                                               
```

@kraj do you recall what the busybox `--install` is for?

kraj, 2023-08-23:
This does install operation on busybox and -s makes symlinks see [1]

fact that --install fails to execute, means that we have to enable this in defconfig for busybox.
maybe we need to add `CONFIG_INSTALL=y` to defconfig. Alternatively we can delete it completely too

[1] https://git.busybox.net/busybox/tree/INSTALL?h=1_34_stable#n104

cbrake, 2023-08-23:
here is the info on that:

```
Installing links to busybox is not always necessary.  The special applet name
"busybox" (or with any optional suffix, such as "busybox-static") uses the
first argument to determine which applet to behave as, for example
"./busybox cat LICENSE".  (Running the busybox applet with no arguments gives
a list of all enabled applets.) The standalone shell can also call busybox
applets without links to busybox under other names in the filesystem.  You can
also configure a standalone install capability into the busybox base applet,
and then install such links at runtime with one of "busybox --install" (for
hardlinks) or "busybox --install -s" (for symlinks).
```

I'm seeing this in the initramfs (below is sampling, there is a lot more):

```
lrwxrwxrwx    1 root     root            23 Mar  9  2018 sort -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 ssl_client -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 stat -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 strings -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 stty -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            21 Mar  9  2018 su -> /usr/bin/busybox.suid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 svc -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 svok -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 sync -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 tail -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 tar -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 tee -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 telnet -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 test -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 tftp -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 time -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 timeout -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 top -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 touch -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 tr -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            21 Mar  9  2018 traceroute -> /usr/bin/busybox.suid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 tree -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 true -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 ts -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 tsort -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 tty -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 umount -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 uname -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 uniq -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 unlink -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 unshare -> /usr/bin/busybox.nosuid
lrwxrwxrwx    1 root     root            23 Mar  9  2018 unzip -> /usr/bin/busybox.nosuid
-rwxr-xr-x    1 root     root          4679 Mar  9  2018 update-alternatives
lrwxrwxrwx    1 root     root            23 Mar  9  2018 uptime -> /usr/bin/busybox.nosuid
```
Can we assume that Yocto has already set up all the links and we don't need to run `busybox --install` or is there something more that this does?

kraj, 2023-08-23:
yeah I would think so. In this case we can just remove the busybox --install -s cmd.

kraj, 2023-09-29:
commit 3ccb819b6a884ef502da36a18bf8ad106b155ad1 fixed it.

closed 2023-09-29 by kraj