YoeDistro/yoe-distro #841: linux-variscite build failure

cbrake, 2023-09-13:
Got this when building recent Yoe for `var-som-mx8x`:

```
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: KBUILD_BUILD_TIMESTAMP: Fri Jul 14 11:33:35 UTC 2023
| NOTE: make -j 24 CC=aarch64-yoe-linux-gcc   -fuse-ld=bfd   -fmacro-prefix-map=/scratch/yoe/yoe-distro/build/tmp/work/imx8qxp_var_som-yoe-linux/linux-variscite/5.15.60+git/git=/usr/src/debug/linux-variscite/5.15.60+git-r0  -fdebug-prefix-map=/scratch/yoe/yoe-distro/build/tmp/work/imx8qxp_var_som-yoe-linux/linux-variscite/5.15.60+git/git=/usr/src/debug/linux-variscite/5.15.60+git-r0  -fmacro-prefix-map=/scratch/yoe/yoe-distro/build/tmp/work/imx8qxp_var_som-yoe-linux/linux-variscite/5.15.60+git/build=/usr/src/debug/linux-variscite/5.15.60+git-r0  -fdebug-prefix-map=/scratch/yoe/yoe-distro/build/tmp/work/imx8qxp_var_som-yoe-linux/linux-variscite/5.15.60+git/build=/usr/src/debug/linux-variscite/5.15.60+git-r0  -fdebug-prefix-map=/scratch/yoe/yoe-distro/build/tmp/work/imx8qxp_var_som-yoe-linux/linux-variscite/5.15.60+git/recipe-sysroot=  -fmacro-prefix-map=/scratch/yoe/yoe-distro/build/tmp/work/imx8qxp_var_som-yoe-linux/linux-variscite/5.15.60+git/recipe-sysroot=  -fdebug-prefix-map=/scratch/yoe/yoe-distro/build/tmp/work/imx8qxp_var_som-yoe-linux/linux-variscite/5.15.60+git/recipe-sysroot-native=  -fdebug-prefix-map=/scratch/yoe/yoe-distro/build/tmp/work-shared/imx8qxp-var-som/kernel-source=/usr/src/kernel -fdebug-prefix-map=/scratch/yoe/yoe-distro/build/tmp/work-shared/imx8qxp-var-som/kernel-build-artifacts=/usr/src/kernel LD=aarch64-yoe-linux-ld.bfd   OBJCOPY=aarch64-yoe-linux-objcopy  STRIP=aarch64-yoe-linux-strip  HOSTCC=gcc  HOSTCFLAGS=-isystem/scratch/yoe/yoe-distro/build/tmp/work/imx8qxp_var_som-yoe-linux/linux-variscite/5.15.60+git/recipe-sysroot-native/usr/include -O2 -pipe HOSTLDFLAGS=-L/scratch/yoe/yoe-distro/build/tmp/work/imx8qxp_var_som-yoe-linux/linux-variscite/5.15.60+git/recipe-sysroot-native/usr/lib                         -L/scratch/yoe/yoe-distro/build/tmp/work/imx8qxp_var_som-yoe-linux/linux-variscite/5.15.60+git/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/scratch/yoe/yoe-distro/build/tmp/work/imx8qxp_var_som-yoe-linux/linux-variscite/5.15.60+git/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/scratch/yoe/yoe-distro/build/tmp/work/imx8qxp_var_som-yoe-linux/linux-variscite/5.15.60+git/recipe-sysroot-native/lib                         -Wl,-rpath,/scratch/yoe/yoe-distro/build/tmp/work/imx8qxp_var_som-yoe-linux/linux-variscite/5.15.60+git/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/scratch/yoe/yoe-distro/build/tmp/work/imx8qxp_var_som-yoe-linux/linux-variscite/5.15.60+git/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/scratch/yoe/yoe-distro/build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 HOSTCPP=gcc  -E HOSTCXX=g++  HOSTCXXFLAGS=-isystem/scratch/yoe/yoe-distro/build/tmp/work/imx8qxp_var_som-yoe-linux/linux-variscite/5.15.60+git/recipe-sysroot-native/usr/include -O2 -pipe -j 24 Image.gz LOADADDR=20008000
| aarch64-yoe-linux-ld.bfd: warning: .tmp_vmlinux.kallsyms1 has a LOAD segment with RWX permissions
| aarch64-yoe-linux-ld.bfd: warning: .tmp_vmlinux.kallsyms2 has a LOAD segment with RWX permissions
| aarch64-yoe-linux-ld.bfd: warning: vmlinux has a LOAD segment with RWX permissions
| gzip: abort: zlib version less than 1.2.3
| make[1]: *** [/scratch/yoe/yoe-distro/build/tmp/work-shared/imx8qxp-var-som/kernel-source/arch/arm64/boot/Makefile:28: arch/arm64/boot/Image.gz] Error 22
| make[1]: *** Deleting file 'arch/arm64/boot/Image.gz'
| make: *** [/scratch/yoe/yoe-distro/build/tmp/work-shared/imx8qxp-var-som/kernel-source/arch/arm64/Makefile:158: Image.gz] Error 2
| ERROR: oe_runmake failed
| WARNING: /scratch/yoe/yoe-distro/build/tmp/work/imx8qxp_var_som-yoe-linux/linux-variscite/5.15.60+git/temp/run.do_compile.1398932:309 exit 1 from 'exit 1'
| WARNING: Backtrace (BB generated script):
```

kraj, 2023-09-14:
| gzip: abort: zlib version less than 1.2.3

I assume build was happening inside docker container ? if so I would suggest to use latest yoebuild container image, inside container we have gzip 1.12 / libz.so.1.2.13 and on my arch box I have 1.13/libz 1.3, so I wonder what might be happening.

cbrake, 2023-09-14:
yes, in the container. I pulled the latest docker image and still got the same thing. I merged the latest Yoe master and trying the i.mx8 nano machine to see if it is happening there.

cbrake, 2023-09-14:
still get the same thing with nano and latest yoe:

| gzip: abort: zlib version less than 1.2.3

cbrake, 2023-09-14:
I ran do_compile manually, and it still fails with same error -- this would be outside the docker container.

cbrake, 2023-09-14:
it is using the following gzip:

`/scratch/yoe/yoe-distro/build/tmp/work/imx8mn_var_som-yoe-linux/linux-variscite/5.15.71+git/recipe-sysroot-native/usr/bin/gzip`

And if I run this manually, I get the error.

kraj, 2023-09-14:
OK that should come from `gzip-native` or `pigz-native` I think but it works here inside container or outside.

```
% ./work/qemux86_64-yoe-linux-musl/meta-oe-ptest-image-mariadb/1.0/recipe-sysroot-native/usr/bin/gzip
Usage: pigz [options] [files ...]
  will compress files in place, adding the suffix '.gz'. If no files are
  specified, stdin will be compressed to stdout. pigz does what gzip does,
  but spreads the work over multiple processors and cores when compressing.

Options:
  -0 to -9, -11        Compression level (level 11, zopfli, is much slower)
  --fast, --best       Compression levels 1 and 9 respectively
  -A, --alias xxx      Use xxx as the name for any --zip entry from stdin
  -b, --blocksize mmm  Set compression block size to mmmK (default 128K)
  -c, --stdout         Write all processed output to stdout (won't delete)
  -C, --comment ccc    Put comment ccc in the gzip or zip header
  -d, --decompress     Decompress the compressed input
  -f, --force          Force overwrite, compress .gz, links, and to terminal
  -F  --first          Do iterations first, before block split for -11
  -h, --help           Display a help screen and quit
  -H, --huffman        Use only Huffman coding for compression
  -i, --independent    Compress blocks independently for damage recovery
  -I, --iterations n   Number of iterations for -11 optimization
  -J, --maxsplits n    Maximum number of split blocks for -11
  -k, --keep           Do not delete original file after processing
  -K, --zip            Compress to PKWare zip (.zip) single entry format
  -l, --list           List the contents of the compressed input
  -L, --license        Display the pigz license and quit
  -m, --no-time        Do not store or restore mod time
  -M, --time           Store or restore mod time
  -n, --no-name        Do not store or restore file name or mod time
  -N, --name           Store or restore file name and mod time
  -O  --oneblock       Do not split into smaller blocks for -11
  -p, --processes n    Allow up to n compression threads (default is the
                       number of online processors, or 8 if unknown)
  -q, --quiet          Print no messages, even on error
  -r, --recursive      Process the contents of all subdirectories
  -R, --rsyncable      Input-determined block locations for rsync
  -S, --suffix .sss    Use suffix .sss instead of .gz (for compression)
  -t, --test           Test the integrity of the compressed input
  -U, --rle            Use run-length encoding for compression
  -v, --verbose        Provide more verbose output
  -V  --version        Show the version of pigz
  -Y  --synchronous    Force output file write to permanent storage
  -z, --zlib           Compress to zlib (.zz) instead of gzip format
  --                   All arguments after "--" are treated as files
```

kraj, 2023-09-14:
Can you run

```
readelf -d gzip
```

cbrake, 2023-09-14:
```
[cbrake@ceres yoe-distro]$ readelf -d /scratch/yoe/yoe-distro/build/tmp/work/imx8mn_var_som-yoe-linux/linux-variscite/5.15.71+git/recipe-sysroot-native/usr/bin/gzip
[

Dynamic section at offset 0x1edb0 contains 29 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libz.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x000000000000001d (RUNPATH)            Library runpath: [$ORIGIN/../lib:$ORIGIN/../../lib]
 0x000000000000000c (INIT)               0x3000
 0x000000000000000d (FINI)               0x18670
 0x0000000000000019 (INIT_ARRAY)         0x1f9b0
 0x000000000000001b (INIT_ARRAYSZ)       8 (bytes)
 0x000000000000001a (FINI_ARRAY)         0x1f9b8
 0x000000000000001c (FINI_ARRAYSZ)       8 (bytes)
 0x000000006ffffef5 (GNU_HASH)           0x3e0
 0x0000000000000005 (STRTAB)             0xda8
 0x0000000000000006 (SYMTAB)             0x418
 0x000000000000000a (STRSZ)              1452 (bytes)
 0x000000000000000b (SYMENT)             24 (bytes)
 0x0000000000000015 (DEBUG)              0x0
 0x0000000000000003 (PLTGOT)             0x1ffe8
 0x0000000000000002 (PLTRELSZ)           2232 (bytes)
 0x0000000000000014 (PLTREL)             RELA
 0x0000000000000017 (JMPREL)             0x22d8
 0x0000000000000007 (RELA)               0x14e0
 0x0000000000000008 (RELASZ)             3576 (bytes)
 0x0000000000000009 (RELAENT)            24 (bytes)
 0x000000006ffffffb (FLAGS_1)            Flags: PIE
 0x000000006ffffffe (VERNEED)            0x1420
 0x000000006fffffff (VERNEEDNUM)         3
 0x000000006ffffff0 (VERSYM)             0x1354
 0x000000006ffffff9 (RELACOUNT)          139
 0x0000000000000000 (NULL)               0x0
```

kraj, 2023-09-29:
@cbrake any more updates on it ? is it still happening?

cbrake, 2023-09-29:
with latest, `bitbake virtual/kernel` now works -- closing ...

closed 2023-09-29 by cbrake