YoeDistro/yoe-distro #847: docker: Error response from daemon: error gathering device information while adding custom device "/dev/kvm": no such file or directory.

mr-nice, 2023-09-29:
Hi all,

I'm building yoe-distro on a virtual server. This worked quite well but after changes to the envsetup.sh I get this error, if I want to build with docker.

docker: Error response from daemon: error gathering device information while adding custom device "/dev/kvm": no such file or directory.

After removing

--device /dev/kvm

from the docker run command I can build fine on that server.

What do you think about a config option to override that behavior? I totally understand that most people build the distribution on real hardware but this comes quite handy if building on virtual servers.

If there is interest in a patch I can submit one.

Greets,
mr-nice

mr-nice, 2023-09-29:
As you can already override docker usage and if you want to use docker or podman, should I just introduce a environment variable for kvm usage aka USE_KVM or should we introduce a config in config/project/container.config where everything is configured? Which just gets consumed by envsetup.sh?  This might also be handy later on to override all docker/container based configurations? It also might be handy to configure the docker uid mapping there? And if you want to run it rootless?

If I understood you right, your preferred patch would be to check if the file exists and is usable? I can do that as well.

kraj, 2023-09-29:
I think if we can make it transparent to user that would be preferred

kraj, 2023-10-05:
this should be fixed now.

closed 2023-10-05 by kraj