YoeDistro/yoe-distro #848: var-som-mx8x imx-boot build failure

cbrake, 2023-09-29:
```
[cbrake@ceres yoe-distro]$ more build/tmp//work/imx8qxp_var_som-yoe-linux/imx-boot/1.0/temp/log.do_compile
DEBUG: Executing shell function do_compile
Copying DTBs
NOTE: 8QX boot binary build
cp: -r not specified; omitting directory '/scratch/yoe/yoe-distro/build/tmp/deploy/images/imx8qxp-var-som/'
WARNING: /scratch/yoe/yoe-distro/build/tmp/work/imx8qxp_var_som-yoe-linux/imx-boot/1.0/temp/run.do_compile.21383:188 exit 1 from 'cp /scratch/yoe/yoe-distro/build/tmp/deploy/images/imx8qxp-var-som/${UBOOT_NAME_EXTRA} /scratch/yoe/yoe-distro/b
uild/tmp/work/imx8qxp_var_som-yoe-linux/imx-boot/1.0/git/iMX8QX/u-boot.bin'
WARNING: Backtrace (BB generated script): 
        #1: compile_mx8x, /scratch/yoe/yoe-distro/build/tmp/work/imx8qxp_var_som-yoe-linux/imx-boot/1.0/temp/run.do_compile.21383, line 188
        #2: do_compile, /scratch/yoe/yoe-distro/build/tmp/work/imx8qxp_var_som-yoe-linux/imx-boot/1.0/temp/run.do_compile.21383, line 154
        #3: main, /scratch/yoe/yoe-distro/build/tmp/work/imx8qxp_var_som-yoe-linux/imx-boot/1.0/temp/run.do_compile.21383, line 196
```

kraj, 2023-09-29:
https://github.com/YoeDistro/meta-variscite-bsp/pull/9

should fix it

cbrake, 2023-09-29:
Thanks, compiles now! Continuing build/testing ...

closed 2023-11-17 by cbrake