YoeDistro/yoe-distro #869: weston does not start on odroid-c4

cbrake, 2023-11-17:
not a critical issue as we don't have any applications that require use of the GPU/display controller, but just FYI in case there is an easy fix.

```
Sep 20 11:23:30 odroid-c4-hardkernel systemd[1]: Started Session c1 of User weston.                                                                                                                                                             
Sep 20 11:23:30 odroid-c4-hardkernel kernel: audit: type=1006 audit(1695209010.072:3): pid=5042 uid=0 old-auid=4294967295 auid=1000 tty=tty7 old-ses=4294967295 ses=2 res=1
Sep 20 11:23:30 odroid-c4-hardkernel kernel[4910]: [  124.783773@1] audit: type=1006 audit(1695209010.072:3): pid=5042 uid=0 old-auid=4294967295 auid=1000 tty=tty7 old-ses=4294967295 ses=2 res=1
Sep 20 11:23:30 odroid-c4-hardkernel weston[5042]: Date: 2023-09-20 UTC                                          
Sep 20 11:23:30 odroid-c4-hardkernel weston[5042]: [11:23:30.112] weston 12.0.2                                                                                                                                                                 
Sep 20 11:23:30 odroid-c4-hardkernel weston[5042]:                https://wayland.freedesktop.org                                                                                                                                               
Sep 20 11:23:30 odroid-c4-hardkernel weston[5042]:                Bug reports to: https://gitlab.freedesktop.org/wayland/weston/issues/          
Sep 20 11:23:30 odroid-c4-hardkernel weston[5042]:                Build: 12.0.2                              
Sep 20 11:23:30 odroid-c4-hardkernel weston[5042]: [11:23:30.112] Command line: /usr/bin/weston --modules=systemd-notify.so                                                                                                                     
Sep 20 11:23:30 odroid-c4-hardkernel weston[5042]: [11:23:30.112] OS: Linux, 4.9.337-odroid, #1 SMP PREEMPT Wed Jul 19 14:26:56 UTC 2023, aarch64
Sep 20 11:23:30 odroid-c4-hardkernel weston[5042]: [11:23:30.112] Flight recorder: enabled                  
Sep 20 11:23:30 odroid-c4-hardkernel weston[5042]: [11:23:30.114] Using config file '/etc/xdg/weston/weston.ini'                                                                                                                                
Sep 20 11:23:30 odroid-c4-hardkernel weston[5042]: [11:23:30.116] Output repaint window is 7 ms maximum.                                                                                                                                        
Sep 20 11:23:30 odroid-c4-hardkernel weston[5042]: [11:23:30.117] Loading module '/usr/lib/libweston-12/drm-backend.so' 
Sep 20 11:23:30 odroid-c4-hardkernel weston[5042]: [11:23:30.122] initializing drm backend                                                                                                                                                      
Sep 20 11:23:30 odroid-c4-hardkernel weston[5042]: [11:23:30.122] Trying libseat launcher...                                                                                                                                                    
Sep 20 11:23:30 odroid-c4-hardkernel weston[5042]: [11:23:30.123] [libseat/backend/seatd.c:64] Could not connect to socket /run/seatd.sock: No such file or directory
Sep 20 11:23:30 odroid-c4-hardkernel weston[5042]: [11:23:30.123] [libseat/libseat.c:76] Backend 'seatd' failed to open seat, skipping                                                                                                          
Sep 20 11:23:30 odroid-c4-hardkernel weston[5042]: [11:23:30.158] [libseat/libseat.c:73] Seat opened with backend 'logind'                              
Sep 20 11:23:30 odroid-c4-hardkernel weston[5042]: [11:23:30.158] libseat: session control granted                                                                                                                                              
Sep 20 11:23:30 odroid-c4-hardkernel weston[5042]: [11:23:30.158] no drm device found                                                                                                                                                           
Sep 20 11:23:30 odroid-c4-hardkernel weston[5042]: [11:23:30.175] fatal: failed to create compositor backend                                                                                                                                    
Sep 20 11:23:30 odroid-c4-hardkernel weston[5042]: Internal warning: debug scope 'drm-backend' has not been destroyed.                                                                                                                          
Sep 20 11:23:30 odroid-c4-hardkernel systemd[1]: weston.service: Main process exited, code=exited, status=1/FAILURE                                                                                                                             
Sep 20 11:23:30 odroid-c4-hardkernel systemd[1]: weston.service: Failed with result 'exit-code'.
Sep 20 11:23:30 odroid-c4-hardkernel systemd[1]: Failed to start Weston, a Wayland compositor, as a system service.
Sep 20 11:23:30 odroid-c4-hardkernel avahi-daemon[4897]: Server startup complete. Host name is odroid-c4-hardkernel.local. Local service cookie is 3112739720.
```

kraj, 2023-11-17:
Can you try with 

```
YOE_PROFILE = "yoe-glibc-systemd-eglfs"
```

cbrake, 2023-11-17:
sure, running ...

I did notice this warning:

[image]

kraj, 2023-11-17:
> sure, running ...
> 
> I did notice this warning:
> 
> [image]

yeah thats ok. We are not doing x11, actually we should perhaps remove that layer from layer.conf for this project.