YoeDistro/yoe-distro #913: qt6.7 compile error

dncex, 2024-03-20:
I'm trying to compile the "yoe-qt6-image" image, for the project "rpi4-32", with the master branch ( 8a65d718dbbb1a79d9c86971986c05a8ece411a2 ).
This is the error I am facing:

`Building CXX object src/qdoc/qdoc/CMakeFiles/qdoc.dir/src/qdoc/clangcodeparser.cpp.o
| FAILED: src/qdoc/qdoc/CMakeFiles/qdoc.dir/src/qdoc/clangcodeparser.cpp.o
| /media/exel/usbdisk/yoe/build/tmp/hosttools/g++ -DCLANG_RESOURCE_DIR=\"/media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/recipe-sysroot-native/usr/lib/clang/18.1.1/include\" -DLIBCLANG_VERSION_MAJOR=18 -DQDOC2_COMPAT -DQT_CORE_LIB -DQT_DISABLE_DEPRECATED_UP_TO=0x050000 -DQT_EXPLICIT_QFILE_CONSTRUCTION_FROM_PATH -DQT_NETWORK_LIB -DQT_NO_AS_CONST -DQT_NO_AS_CONST=1 -DQT_NO_CONTEXTLESS_CONNECT=1 -DQT_NO_DEBUG -DQT_NO_EXCEPTIONS -DQT_NO_FOREACH -DQT_NO_FOREACH=1 -DQT_NO_JAVA_STYLE_ITERATORS -DQT_NO_NARROWING_CONVERSIONS_IN_CONNECT -DQT_NO_QEXCHANGE -DQT_QMLBUILTINS_LIB -DQT_QMLINTEGRATION_LIB -DQT_QML_LIB -DQT_USE_QSTRINGBUILDER -DQT_WARN_DEPRECATED_UP_TO=0x070000 -D_LARGEFILE64_SOURCE -D_LARGEFILE_SOURCE -I/media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/build/src/qdoc/qdoc/qdoc_autogen/include -I/media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/git/src/qdoc/qdoc -I/media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/build/src/qdoc/qdoc -I/media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/git/src/qdoc/qdoc/src -I/media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/recipe-sysroot-native/usr/include/QtQml/6.7.0 -I/media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/recipe-sysroot-native/usr/include/QtQml/6.7.0/QtQml -I/media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/recipe-sysroot-native/usr/include/QtCore/6.7.0 -I/media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/recipe-sysroot-native/usr/include/QtCore/6.7.0/QtCore -I/media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/recipe-sysroot-native/usr/include/QtCore -I/media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/recipe-sysroot-native/usr/lib/mkspecs/linux-g++ -I/media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/recipe-sysroot-native/usr/include/QtQmlBuiltins/6.7.0 -I/media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/recipe-sysroot-native/usr/include/QtQmlBuiltins/6.7.0/QtQmlBuiltins -I/media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/recipe-sysroot-native/usr/include/QtQmlBuiltins -I/media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/recipe-sysroot-native/usr/include/QtQml -I/media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/recipe-sysroot-native/usr/include/QtQmlIntegration -I/media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/recipe-sysroot-native/usr/include/QtNetwork -isystem/media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/recipe-sysroot-native/usr/include -O2 -pipe -DNDEBUG -O2 -std=c++17 -fvisibility=hidden -fvisibility-inlines-hidden -Wall -Wextra -fno-exceptions -fPIC -Wsuggest-override -MD -MT src/qdoc/qdoc/CMakeFiles/qdoc.dir/src/qdoc/clangcodeparser.cpp.o -MF src/qdoc/qdoc/CMakeFiles/qdoc.dir/src/qdoc/clangcodeparser.cpp.o.d -o src/qdoc/qdoc/CMakeFiles/qdoc.dir/src/qdoc/clangcodeparser.cpp.o -c /media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/git/src/qdoc/qdoc/src/qdoc/clangcodeparser.cpp
| In file included from /media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/git/src/qdoc/qdoc/src/qdoc/clangcodeparser.cpp:40:
| /media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/git/src/qdoc/qdoc/src/qdoc/clang/AST/QualTypeNames.h: In function ‘clang::QualType clang::TypeName::getFullyQualifiedType(clang::QualType, const clang::ASTContext&, bool)’:
| /media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/git/src/qdoc/qdoc/src/qdoc/clang/AST/QualTypeNames.h:441:35: error: ‘ETK_None’ was not declared in this scope; did you mean ‘EST_None’?
|   441 |   ElaboratedTypeKeyword Keyword = ETK_None;
|       |                                   ^~~~~~~~
|       |                                   EST_None
| [646/708] Building CXX object src/qdoc/qdoc/CMakeFiles/qdoc.dir/src/qdoc/enumnode.cpp.o
| [647/708] Building CXX object src/qdoc/qdoc/CMakeFiles/qdoc.dir/src/qdoc/functionnode.cpp.o
| [648/708] Building CXX object src/qdoc/qdoc/CMakeFiles/qdoc.dir/src/qdoc/docparser.cpp.o
| [649/708] Building CXX object src/qdoc/qdoc/CMakeFiles/qdoc.dir/src/qdoc/headernode.cpp.o
| [650/708] Building CXX object src/qdoc/qdoc/CMakeFiles/qdoc.dir/src/qdoc/docbookgenerator.cpp.o
| [651/708] Building CXX object src/qdoc/qdoc/CMakeFiles/qdoc.dir/src/qdoc/helpprojectwriter.cpp.o
| [652/708] Building CXX object src/qdoc/qdoc/CMakeFiles/qdoc.dir/src/qdoc/generator.cpp.o
| ninja: build stopped: subcommand failed.
| WARNING: /media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/temp/run.do_compile.30741:160 exit 1 from 'eval ${DESTDIR:+DESTDIR=${DESTDIR} } cmake --build '/media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/build' "$@" -- ${EXTRA_OECMAKE_BUILD}'
| WARNING: Backtrace (BB generated script):
| 	#1: cmake_runcmake_build, /media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/temp/run.do_compile.30741, line 160
| 	#2: cmake_do_compile, /media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/temp/run.do_compile.30741, line 154
| 	#3: do_compile, /media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/temp/run.do_compile.30741, line 149
| 	#4: main, /media/exel/usbdisk/yoe/build/tmp/work/x86_64-linux/qttools-native/6.7.1/temp/run.do_compile.30741, line 173
ERROR: Task (virtual:native:/media/exel/usbdisk/yoe/sources/meta-qt6/recipes-qt/qt6/qttools_git.bb:do_compile) failed with exit code '1'`

BTW: the compile fails also with the tag "2024.02", but for different reason:

`ERROR: qtquick3dphysics-6.6.3-r0 do_patch: Applying patch '0001-PhysX-Check-for-SSE-to-define-__intXX_t.patch' on target directory '/media/exel/usbdisk/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/qtquick3dphysics/6.6.3/git'
CmdError('quilt --quiltrc /media/exel/usbdisk/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/qtquick3dphysics/6.6.3/recipe-sysroot-native/etc/quiltrc push', 0, 'stdout: Applying patch 0001-PhysX-Check-for-SSE-to-define-__intXX_t.patch
The next patch would create the file src/3rdparty/PhysX/0001-PhysX-Check-for-SSE-to-define-__intXX_t.patch,
which already exists!  Applying it anyway.
patching file src/3rdparty/PhysX/0001-PhysX-Check-for-SSE-to-define-__intXX_t.patch
Hunk #1 FAILED at 1.
1 out of 1 hunk FAILED -- rejects in file src/3rdparty/PhysX/0001-PhysX-Check-for-SSE-to-define-__intXX_t.patch
patching file src/3rdparty/PhysX/source/foundation/include/unix/sse2/PsUnixSse2AoS.h
Hunk #1 FAILED at 43.
1 out of 1 hunk FAILED -- rejects in file src/3rdparty/PhysX/source/foundation/include/unix/sse2/PsUnixSse2AoS.h
Patch 0001-PhysX-Check-for-SSE-to-define-__intXX_t.patch can be reverse-applied

stderr: ')
ERROR: Logfile of failure stored in: /media/exel/usbdisk/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/qtquick3dphysics/6.6.3/temp/log.do_patch.656917
ERROR: Task (/media/exel/usbdisk/yoe/sources/meta-qt6/recipes-qt/qt6/qtquick3dphysics_git.bb:do_patch) failed with exit code '1'
`

kraj, 2024-03-20:
@dncex thanks for report !, I think you are running into this issue 

https://bugreports.qt.io/browse/QTBUG-123109

good news is that its fixed in qttools by

https://github.com/qt/qttools/commit/a916ac4087b1a0ec76b7fb5b3c003274d685716d

This has made into upstream meta-qt6 and with 

https://github.com/YoeDistro/meta-qt6/pull/39

is being pulled into yoe fork. Once its merged, we will bump the submodules in yoedistro and finally the fix
will be available in 2024.03 release. It might take few days before it shows up in `master` branch of yoedistro

Meanwhile you can apply the above patch manually to qttools in your checkout and see if this helps.

Yoe CI build uses `yoe-kiosk-image` which is based on qtwebengine and does not use/excercise all the components of QT project

kraj, 2024-03-21:
https://github.com/YoeDistro/yoe-distro/pull/914 should fix it

dncex, 2024-03-21:
thanks for the prompt feedback, qttools compile is ok with the suggested patch ( https://github.com/qt/qttools/commit/a916ac4087b1a0ec76b7fb5b3c003274d685716d ).

Anyhow I see another error on the package gstreamer1.0-plugins-bad (I suppose it is a dependecy of qtmultimedia):
`
| Subproject  libva is buildable: NO (disabling)
| Dependency libva from subproject libva found: NO (subproject failed to configure)
| Run-time dependency libva-drm found: NO (tried pkgconfig and cmake)
| Looking for a fallback subproject for the dependency libva-drm
| Dependency libva-drm from subproject libva found: NO (subproject failed to configure)
| Run-time dependency vulkan found: YES 1.3.275
| Has header "vulkan/vulkan_core.h" : YES
| Dependency xcb skipped: feature x11 disabled
| Dependency xkbcommon skipped: feature x11 disabled
| Dependency xkbcommon-x11 skipped: feature x11 disabled
| Dependency wayland-client skipped: feature wayland disabled
| 
| ../gst-plugins-bad-1.22.10/gst-libs/gst/vulkan/meson.build:269:4: ERROR: Problem encountered: No Windowing system found. vulkansink will not work
| 
| A full log can be found at /media/exel/usbdisk/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/gstreamer1.0-plugins-bad/1.22.10/build/meson-logs/meson-log.txt
| ERROR: meson failed
| WARNING: /media/exel/usbdisk/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/gstreamer1.0-plugins-bad/1.22.10/temp/run.do_configure.1014:175 exit 1 from 'exit 1'
| WARNING: Backtrace (BB generated script):
| 	#1: bbfatal_log, /media/exel/usbdisk/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/gstreamer1.0-plugins-bad/1.22.10/temp/run.do_configure.1014, line 175
| 	#2: meson_do_configure, /media/exel/usbdisk/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/gstreamer1.0-plugins-bad/1.22.10/temp/run.do_configure.1014, line 164
| 	#3: do_configure, /media/exel/usbdisk/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/gstreamer1.0-plugins-bad/1.22.10/temp/run.do_configure.1014, line 153
| 	#4: main, /media/exel/usbdisk/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/gstreamer1.0-plugins-bad/1.22.10/temp/run.do_configure.1014, line 188
ERROR: Task (/media/exel/usbdisk/yoe/sources/poky/meta/recipes-multimedia/gstreamer/gstreamer1.0-plugins-bad_1.22.10.bb:do_configure) failed with exit code '1'
`

kraj, 2024-03-21:
Does it build ok when you use `YOE_PROFILE = "yoe-glibc-systemd-wayland"`

dncex, 2024-03-22:
actually reverting this patch the compile will go:
https://gitlab.freedesktop.org/gstreamer/gst-plugins-bad/-/merge_requests/1391/diffs?commit_id=d9c5b5cb938e0d4f5daff174a95037ab7590bbe3
I will test it and let you know