YoeDistro/yoe-distro #938: ca-certificates missing from latest build?

cbrake, 2024-05-20:
In the latest rPI builds, it seems ca-certificates is missing from the image:

```
root@raspberrypi4-64:~# wget https://files.bec-systems.com/yoe/files.txt
Connecting to files.bec-systems.com (165.227.252.84:443)
wget: note: TLS certificate validation not implemented
wget: TLS error from peer (alert code 40): handshake failure
wget: error getting response: Connection reset by peer

opkg list_installed  | grep cert
(nothing)
```

I checked in an older project build, and it is included:

```
opkg list_installed | grep cert
ca-certificates - 20211016-r0.0
```

@kraj are you seeing the same thing, and if so, where is the best place to add it back in?

kraj, 2024-05-20:
lets add it to `IMAGE_INSTALL` intentionally via
`meta-yoe/recipes-core/images/yoe-simple-image.bb`

cbrake, 2024-05-20:
queued in PR

closed 2024-05-20 by cbrake