YoeDistro/yoe-distro #940: kiosk browser is not running correctly on rPI4/Weston

cbrake, 2024-05-20:
The app starts, and I see the QML loading screen, but the webengine screen never loads.

Trying various other combinations of platforms ...

log:

```
YOE_KIOSK_BROWSER_URL= "https://yoedistro.org"
YOE_KIOSK_BROWSER_EXCEPTION_URL= ""
YOE_KIOSK_BROWSER_ROTATE= "0"
YOE_KIOSK_BROWSER_KEYBOARD_SCALE= "1"
YOE_KIOSK_BROWSER_RETRY_INTERVAL= "5"
qt.qpa.wayland: qtvirtualkeyboard currently is not supported at client-side, use QT_IM_MODULE=qtvirtualkeyboard at compositor-side.
Path override failed for key base::DIR_APP_DICTIONARIES and path '/usr/bin/qtwebengine_dictionaries'
Path override failed for key base::DIR_APP_DICTIONARIES and path '/usr/libexec/qtwebengine_dictionaries'
Path override failed for key base::DIR_APP_DICTIONARIES and path '/usr/libexec/qtwebengine_dictionaries'
Attempting to create QWindow-based QOffscreenSurface outside the gui thread. Expect failures.
Attempting to create QWindow-based QOffscreenSurface outside the gui thread. Expect failures.
Attempting to create QWindow-based QOffscreenSurface outside the gui thread. Expect failures.
Attempting to create QWindow-based QOffscreenSurface outside the gui thread. Expect failures.
Attempting to create QWindow-based QOffscreenSurface outside the gui thread. Expect failures.
Attempting to create QWindow-based QOffscreenSurface outside the gui thread. Expect failures.
Attempting to create QWindow-based QOffscreenSurface outside the gui thread. Expect failures.
Attempting to create QWindow-based QOffscreenSurface outside the gui thread. Expect failures.
Attempting to create QWindow-based QOffscreenSurface outside the gui thread. Expect failures.
Attempting to create QWindow-based QOffscreenSurface outside the gui thread. Expect failures.
```

cbrake, 2024-05-20:
seeing same thing on rPI EGLFS -- trying imx8 nano ...

kraj, 2024-05-21:
> seeing same thing on rPI EGLFS -- trying imx8 nano ...

I can see something similar, where a white window appears but nothing gets creared in it. I wonder if you see same issue on imx based system as well, this could mean its a common problem.

cbrake, 2024-05-21:
kiosk-browser works fine on mx8m-nano, so appears to be a Qt webengine/rPI issue

kraj, 2024-05-21:
> kiosk-browser works fine on mx8m-nano, so appears to be a Qt webengine/rPI issue

RPI uses upstream mesa for graphics and its likely that its broken there somehow or its specific to webengine/chromium
that does not work for ISA used in rpi ( maybe miscompiled ), I will try to look a bit more.

cbrake, 2024-05-31:
kiosk-browser works fine on v24.3.0. Uploaded to:

https://files.bec-systems.com/yoe/testing/

cbrake, 2024-05-31:
v24.5.0 from the Github Yoe release works :-)

[image]

JohanDevelon, 2025-08-05:
Hello,

I try to build kiosk project, but after many tests, I'm not able to a release.
one error it same than first message.
It's not possible to build under 32 bits. qtwebengine failed.
under 64 bits, browser not start.

64 bits + wayland 	=> wayland desktop but no browser:  qt.qpa.wayland: qtvirtualkeyboard currently is not supported at client-side
64 bits + egfls 	=> balck screen : Failed to initialize Vulkan

64 bits + wayland 	=> build error on qtwebengine - native_skia_output_device_opengl.cpp
32 bits + egfls 	=> build error on qtwebengine - native_skia_output_device_opengl.cpp

I Have a Server with 16CPU, 64Go RAM ECC, 500Go Nvme.
I install fresh debian 12.11 Bookworm release. Fully upgraded.
I have SwapFile of 16Go.
I install only Docker. Docker run hello-world is OK.
I want to create a kiosk with an RPI 4B v1.2 + 4Go RAM + Touchscreen 7" (whareshare) connected by DSI connector.

Find bellow, my procedures and logs.
Is there something I make wrong ? 

***********************
I follow procedure
git clone --recurse-submodules -j8 -b master https://github.com/YoeDistro/yoe-distro.git yoe
cd yoe
. ./envsetup.sh rpi4-64
yoe_setup
bitbake yoe-simple-image
=> OK image builded
=> OK image working
*******************

Now, try kiosk image. I customize nothing, I don't create local.config, just default values. 
bitbake yoe-kiosk-image
=> OK image builded
=> weston "desktop" not visible on DSI screen.
=> OK I connect HDMI cable to be sure: I see weston/wayland desktop alone.
=> NO chromium window

logs are:
	The job identifier is 2138.
	Aug 04 21:31:11 raspberrypi4-64 yoe-kiosk-browser[1109]: YOE_KIOSK_BROWSER_URL= "http://localhost:8118"
	Aug 04 21:31:11 raspberrypi4-64 yoe-kiosk-browser[1109]: YOE_KIOSK_BROWSER_EXCEPTION_URL= ""
	Aug 04 21:31:11 raspberrypi4-64 yoe-kiosk-browser[1109]: YOE_KIOSK_BROWSER_ROTATE= "0"
	Aug 04 21:31:11 raspberrypi4-64 yoe-kiosk-browser[1109]: YOE_KIOSK_BROWSER_KEYBOARD_SCALE= "1"
	Aug 04 21:31:11 raspberrypi4-64 yoe-kiosk-browser[1109]: YOE_KIOSK_BROWSER_DEFAULT_DIALOGS= "0"
	Aug 04 21:31:11 raspberrypi4-64 yoe-kiosk-browser[1109]: YOE_KIOSK_BROWSER_DIALOG_COLOR= "#D91824"
	Aug 04 21:31:11 raspberrypi4-64 yoe-kiosk-browser[1109]: YOE_KIOSK_BROWSER_RETRY_INTERVAL= "10"
	Aug 04 21:31:11 raspberrypi4-64 yoe-kiosk-browser[1109]: YOE_KIOSK_BROWSER_FULLSCREEN= "1"
	Aug 04 21:31:11 raspberrypi4-64 yoe-kiosk-browser[1109]: YOE_KIOSK_BROWSER_TOUCH_QUIRK= "1"
	Aug 04 21:31:11 raspberrypi4-64 yoe-kiosk-browser[1109]: YOE_KIOSK_BROWSER_IGNORE_CERT_ERR= "1"
	Aug 04 21:31:11 raspberrypi4-64 yoe-kiosk-browser[1109]: qt.qpa.wayland: qtvirtualkeyboard currently is not supported at client-side, use QT_IM_MODULES=qtvirtualkeyboard at compositor-side.
	Aug 04 21:31:11 raspberrypi4-64 yoe-kiosk-browser[1109]: Sandboxing disabled by user.
	Aug 04 21:31:12 raspberrypi4-64 audit[1109]: ANOM_ABEND auid=4294967295 uid=1000 gid=1000 ses=4294967295 pid=1109 comm="yoe-kiosk-brows" exe="/usr/bin/yoe-kiosk-browser" sig=4 res=1
	Aug 04 21:31:12 raspberrypi4-64 systemd-coredump[1138]: Process 1109 (yoe-kiosk-brows) of user 1000 terminated abnormally with signal 4/ILL, processing...
	Aug 04 21:31:12 raspberrypi4-64 audit: BPF prog-id=64 op=LOAD
	Aug 04 21:31:12 raspberrypi4-64 audit: BPF prog-id=65 op=LOAD
	Aug 04 21:31:12 raspberrypi4-64 audit: BPF prog-id=66 op=LOAD
	Aug 04 21:31:12 raspberrypi4-64 systemd[1]: Started Process Core Dump (PID 1138/UID 0).
	 Subject: A start job for unit systemd-coredump@13-1138-0.service has finished successfully
	 Defined-By: systemd
	 Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel

I try with no DSI screen -> replaced by HDMI, I try with and whitout external keyboard/mouse. Same results.

If I start manually the kiosk program, an other error appear before previous log:
	Detected locale "C" with character encoding "ANSI_X3.4-1968", which is not UTF-8.
	Qt depends on a UTF-8 locale, and has switched to "C.UTF-8" instead.
I set locale to en_US.UTF8. same result.
[... 101 lines cut ...]
|       |                                               static_cast<EGLAttrib>(        )
| 14 warnings and 1 error generated.

	| ninja: build stopped: subcommand failed.
| [89/337] Linking CXX shared module lib/qml/QtWebEngine/ControlsDelegates/libqtwebenginequickdelegatesplugin.so
| FAILED: [code=1] src/core/Release/arm/QtWebEngineCore.stamp src/core/Release/arm/QtWebEngineCore /home/jdevelon/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/qtwebengine/6.9.2/build/src/core/Release/arm/QtWebEngineCore.stamp /home/jdevelon/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/qtwebengine/6.9.2/build/src/core/Release/arm/QtWebEngineCore
| cd /home/jdevelon/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/qtwebengine/6.9.2/build/src/core && /home/jdevelon/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/qtwebengine/6.9.2/recipe-sysroot-native/usr/bin/ninja -C /home/jdevelon/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/qtwebengine/6.9.2/build/src/core/Release/arm QtWebEngineCore
| ninja: build stopped: subcommand failed.
| WARNING: /home/jdevelon/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/qtwebengine/6.9.2/temp/run.do_compile.2145454:159 exit 1 from 'eval ${DESTDIR:+DESTDIR=${DESTDIR} } cmake --build '/home/jdevelon/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/qtwebengine/6.9.2/build' "$@" -- ${EXTRA_OECMAKE_BUILD}'
| WARNING: Backtrace (BB generated script):
[... cut ...]
| 	#4: main, /home/jdevelon/yoe/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/qtwebengine/6.9.2/temp/run.do_compile.2145454, line 172
NOTE: recipe qtwebengine-6.9.2-r0: task do_compile: Failed
ERROR: Task (/home/jdevelon/yoe/sources/meta-qt6/recipes-qt/qt6/qtwebengine_git.bb:do_compile) failed with exit code '1'

*****************
Now, I move to rpi-32 and variable YOE_PROFILE = "yoe-glibc-systemd-wayland" from local.config to go back to wayland
. ./envsetup.sh rpi4-32
yoe_setup
bitbake yoe-kiosk-image

exactly same error, same lines than rpi-32 and variable YOE_PROFILE = "yoe-glibc-systemd-egfls" 
****************

kraj, 2025-08-05:
@JohanDevelon thanks for trying yoe distro and welcome here, I see that you are trying 32bit pi4, we do not test 32bit Pi as much as we test pi4 64bit, so perhaps trying with pi4 64bit might be a good option. Secondly, are you using latest tip if tree if not please try either the monthly release tag or tip if trunk.

kraj, 2025-08-05:
it seems that weston failed to start. what does

```
systemctl --failed
```

tell you ?

kraj, 2025-08-08:
I am able to see crash after few seconds of booting, infact with latest yoe/master on wayland as well. 

I grew curiout about this in your logs

```
terminated abnormally with signal 4/ILL,
```

Please try running
```
bitbake-getvar TUNE_CCARGS -r qtwebengine
```

and see what is says, My hunch is +nocrypto is missing from -mcpu argument.

JohanDevelon, 2025-08-09:
Sorry, for previous wrong logs .

The requested log :
```
bitbake-getvar TUNE_CCARGS -r qtwebengine

#
# $TUNE_CCARGS [28 operations]
#   :remove[toolchain-clang] /home/jdevelon/yoe-july/sources/meta-clang/conf/nonclangable.conf:2
#     "${@bb.utils.contains('TUNE_FEATURES', 'aarch64', '-mbranch-protection=standard', '', d)}"
#   :remove[pn-systemd:toolchain-clang] /home/jdevelon/yoe-july/sources/meta-clang/conf/nonclangable.conf:335
#     "-mfpmath=sse"
#   :remove[pn-systemd-boot:toolchain-clang] /home/jdevelon/yoe-july/sources/meta-clang/conf/nonclangable.conf:336
#     "-mfpmath=sse"
#   :remove[pn-omxplayer:toolchain-clang] /home/jdevelon/yoe-july/sources/meta-clang/conf/nonclangable.conf:338
#     "-no-integrated-as"
#   :remove[pn-nfs-utils:toolchain-clang] /home/jdevelon/yoe-july/sources/meta-clang/conf/nonclangable.conf:339
#     "-Qunused-arguments"
#   :remove[pn-pipewire:toolchain-clang] /home/jdevelon/yoe-july/sources/meta-clang/conf/nonclangable.conf:342
#     "-Qunused-arguments"
#   :remove[pn-tesseract:toolchain-clang] /home/jdevelon/yoe-july/sources/meta-clang/conf/nonclangable.conf:343
#     "-Qunused-arguments"
#   :remove[pn-pulseaudio:toolchain-clang] /home/jdevelon/yoe-july/sources/meta-clang/conf/nonclangable.conf:344
#     "-Qunused-arguments"
#   :remove[pn-btrfs-tools:toolchain-clang] /home/jdevelon/yoe-july/sources/meta-clang/conf/nonclangable.conf:345
#     "-Qunused-arguments"
#   :append[pn-btrfs-tools:toolchain-clang] /home/jdevelon/yoe-july/sources/meta-clang/conf/nonclangable.conf:346
#     " -Werror=unused-command-line-argument"
#   :remove[pn-gimp:toolchain-clang] /home/jdevelon/yoe-july/sources/meta-clang/conf/nonclangable.conf:347
#     "-Qunused-arguments"
#   :append[pn-ceres-solver:toolchain-clang:powerpc] /home/jdevelon/yoe-july/sources/meta-clang/conf/nonclangable.conf:354
#     " -mno-altivec"
#   :append[pn-opencv:toolchain-clang:powerpc] /home/jdevelon/yoe-july/sources/meta-clang/conf/nonclangable.conf:355
#     " -mno-altivec"
#   :remove[pn-kernel-selftest:toolchain-clang] /home/jdevelon/yoe-july/sources/meta-clang/conf/nonclangable.conf:696
#     "-mfpmath=sse"
#   set /home/jdevelon/yoe-july/sources/poky/meta/conf/bitbake.conf:151
#     [_defaultval] ""
#   set /home/jdevelon/yoe-july/sources/poky/meta/conf/bitbake.conf:152
#     [vardepvalue] "${TUNE_CCARGS}"
#   postdot /home/jdevelon/yoe-july/sources/poky/meta/conf/machine/include/arm/armv8a/tune-cortexa72.inc:4
#     "${@bb.utils.contains('TUNE_FEATURES', 'cortexa72', ' -mcpu=cortex-a72', '', d)}"
#   postdot /home/jdevelon/yoe-july/sources/poky/meta/conf/machine/include/arm/arch-arm.inc:20
#     "${TUNE_CCARGS_MARCH}${TUNE_CCARGS_MARCH_OPTS}"
#   postdot /home/jdevelon/yoe-july/sources/poky/meta/conf/machine/include/arm/feature-arm-thumb.inc:22
#     "${@bb.utils.contains('TUNE_FEATURES', 'thumb', ' -m${ARM_M_OPT}', '', d)}"
#   postdot /home/jdevelon/yoe-july/sources/poky/meta/conf/machine/include/arm/feature-arm-vfp.inc:9
#     "${@ (' -mfpu=%s' % d.getVar('TUNE_CCARGS_MFPU').split()[-1]) if (d.getVar('TUNE_CCARGS_MFPU') != '') else ''}"
#   postdot /home/jdevelon/yoe-july/sources/poky/meta/conf/machine/include/arm/feature-arm-vfp.inc:15
#     "${@ ' -mfloat-abi=${TUNE_CCARGS_MFLOAT}' if (d.getVar('TUNE_CCARGS_MFLOAT') != '')     else ''}"
#   postdot /home/jdevelon/yoe-july/sources/poky/meta/conf/machine/include/arm/arch-arm64.inc:44
#     "${@bb.utils.contains('TUNE_FEATURES', 'aarch64', ' -mbranch-protection=standard', '', d)}"
#   :remove[toolchain-clang] /home/jdevelon/yoe-july/sources/meta-clang/classes/clang-legacy.bbclass:11
#     "-mcpu=octeontx2${TUNE_CCARGS_MARCH_OPTS}"
#   :remove[toolchain-clang:powerpc] /home/jdevelon/yoe-july/sources/meta-clang/classes/clang-legacy.bbclass:14
#     "-mhard-float -mno-spe"
#   :append[toolchain-clang:libc-musl:powerpc64] /home/jdevelon/yoe-july/sources/meta-clang/classes/clang-legacy.bbclass:15
#     " -mlong-double-64"
#   :append[toolchain-clang:libc-musl:powerpc64le] /home/jdevelon/yoe-july/sources/meta-clang/classes/clang-legacy.bbclass:16
#     " -mlong-double-64"
#   :append[toolchain-clang:libc-musl:powerpc] /home/jdevelon/yoe-july/sources/meta-clang/classes/clang-legacy.bbclass:17
#     " -mlong-double-64"
#   append /home/jdevelon/yoe-july/sources/poky/meta/classes/toolchain/clang.bbclass:32
#     "${@bb.utils.contains("DISTRO_FEATURES", "usrmerge", " --dyld-prefix=/usr", "", d)}"
# pre-expansion value:
#   "${@bb.utils.contains('TUNE_FEATURES', 'cortexa72', ' -mcpu=cortex-a72', '', d)}${TUNE_CCARGS_MARCH}${TUNE_CCARGS_MARCH_OPTS}${@bb.utils.contains('TUNE_FEATURES', 'thumb', ' -m${ARM_M_OPT}', '', d)}${@ (' -mfpu=%s' % d.getVar('TUNE_CCARGS_MFPU').split()[-1]) if (d.getVar('TUNE_CCARGS_MFPU') != '') else ''}${@ ' -mfloat-abi=${TUNE_CCARGS_MFLOAT}' if (d.getVar('TUNE_CCARGS_MFLOAT') != '')     else ''}${@bb.utils.contains('TUNE_FEATURES', 'aarch64', ' -mbranch-protection=standard', '', d)} ${@bb.utils.contains("DISTRO_FEATURES", "usrmerge", " --dyld-prefix=/usr", "", d)}"
TUNE_CCARGS=" -mcpu=cortex-a72+crc   --dyld-prefix=/usr"

```

kraj, 2025-08-09:
> Sorry, for previous wrong logs .
>
> The requested log :
>
> ```
> bitbake-getvar TUNE_CCARGS -r qtwebengine
>
> #
> # $TUNE_CCARGS [28 operations]
> #   :remove[toolchain-clang] [...]

OK there is one problem here which is that it should have been `-mcpu=cortex-a72+crc+nocrypto` and the reason is that we are missing this commit in oe-core

https://lore.kernel.org/openembedded-core/20250806050838.3311037-1-raj.khem@gmail.com/T/#u

Right now there is a bit of churn since clang compiler which we use in yoe, is moving from meta-clang to oe-core layer.
its not usual but one time situation.

I would suggest that you try `yoe/mut` branch of yoe-distro since it has alll needed commits which are in flight for things to work correctly. I am sure it will all settle by 2025.08 release.