YoeDistro/yoe-distro #942: mx8-nano bluetooth package issues

cbrake, 2024-05-21:
```
ERROR: yoe-kiosk-image-1.0-r0 do_rootfs: Unable to install packages. Command '/scratch4/yoe/yoe-distro/build/tmp/work/imx8mn_var_som-yoe-linux/yoe-kiosk-image/1.0/recipe-sysroot-native/usr/bin/opkg --volatile-cache -f /scratch4/yoe/yoe-distro/build/tmp/work/imx8mn_var_som-yoe-linux/yoe-kiosk-image/1.0/opkg.conf -t /scratch4/yoe/yoe-distro/build/tmp/work/imx8mn_var_som-yoe-linux/yoe-kiosk-image/1.0/temp/ipktemp/ -o /scratch4/yoe/yoe-distro/build/tmp/work/imx8mn_var_som-yoe-linux/yoe-kiosk-image/1.0/rootfs  --force_postinstall --prefer-arch-to-version   --add-ignore-recommends eudev-hwdb --add-ignore-recommends libmodule-build-perl --add-ignore-recommends udev-hwdb install ca-certificates htop kernel-devicetree kernel-image liberation-fonts networkmanager opkg os-release packagegroup-base-ipv6 packagegroup-base-wifi packagegroup-base-zeroconf packagegroup-core-boot packagegroup-core-ssh-openssh packagegroup-distro-base packagegroup-machine-base psplash qtvirtualkeyboard run-postinsts simpleiot wireguard-tools yoe-kiosk-browser' returned 1:
 * Solver encountered 1 problem(s):
 * Problem 1/1:
 *   - package packagegroup-tools-bluetooth-1.0-r0.0.all requires bluez5-testtools, but none of the providers can be installed
 *   - package packagegroup-machine-base-1.0-r0.0.imx8mn_var_som requires packagegroup-tools-bluetooth, but none of the providers can be installed
 *   - package bluez5-testtools-5.72-r0.0.cortexa53-crypto requires python3-pygobject, but none of the providers can be installed
 *   - conflicting requests
 *   - nothing provides python3-pycairo needed by python3-pygobject-3.48.2-r0.0.cortexa53-crypto-mx8mn
 *
 * Solution 1:
 *   - do not ask to install a package providing packagegroup-machine-base

Disfavor package: udev-hwdb

ERROR: Logfile of failure stored in: /scratch4/yoe/yoe-distro/build/tmp/work/imx8mn_var_som-yoe-linux/yoe-kiosk-image/1.0/temp/log.do_rootfs.9049
ERROR: Task (/scratch4/yoe/yoe-distro/sources/meta-yoe/recipes-core/images/yoe-kiosk-image.bb:do_rootfs) failed with exit code '1'
```

cbrake, 2024-05-21:
an older build has the following installed:

```
opkg list_installed | grep blue
bluealsa - 4.0.0+git0+dd05a1baa2-r0.0
bluealsa-aplay - 4.0.0+git0+dd05a1baa2-r0.0
bluez5 - 5.72-r0.0
bluez5-noinst-tools - 5.72-r0.0
bluez5-obex - 5.72-r0.0
bluez5-testtools - 5.72-r0.0
packagegroup-tools-bluetooth - 1.0-r0.0
```

cbrake, 2024-05-21:
`ipk/cortexa53-crypto/bluez5-testtools_5.72-r0.0_cortexa53-crypto.ipk` exists

kraj, 2024-05-21:
does it help if you do 
```
bitbake -ccleansstate python3-pygobject
bitbake yoe-kiosk-image
```

My hunch is that its the mixing of eglfs and wayland based distros in same sstate cache

cbrake, 2024-05-21:
no, that did not fix it. I'll try a clean build.

cbrake, 2024-05-21:
full rebuild worked, closing.

closed 2024-05-21 by cbrake

kraj, 2024-05-21:
> full rebuild worked, closing.

I think this issue is duplicate of https://github.com/YoeDistro/yoe-distro/issues/820
which we have triaged a bit more and still needs to be fixed.

cbrake, 2024-05-21:
good recall! So looks like I can keep sstate in the future and just remove build/tmp to work around this.