YoeDistro/yoe-distro #950: Generic/Intel x86_64 support

poett1, 2024-06-25:
Hello,

I noticed that this distro previously supported x86_64 CPUs in older branches. The meta-intel layer is still present in the submodules; however, the configuration files for these CPUs seem to be missing in conf/projects. Are there any plans to reintroduce support for x86_64 architectures in the near future?

Thanks a lot!

kraj, 2024-06-25:
@poett1 thanks for bringing this up and trying yoe distro !!, our default projects https://github.com/YoeDistro/yoe-distro/tree/master/conf/projects is based on what someone provided and maintains, for x86 we have qemux86 and qemux86-64 projects which are fairly well tested but we do not have projects supporting machines from meta-intel. We can add that if some contributions were made.

granitepeak99, 2024-08-25:
I'm hoping to use yoe distro soon for the Intel NUC to replace a home grown arch linux based distro.  I'm new to Yocto but would love to learn how to help add x86_64 back into yoe.